mrene/minidsp-rs #20: PEQ input import, clear, and bypass do not work

linerjoe, 2021-01-11:
PEQ input import, clear, and bypass seem to work (no errors), but do not actually apply. 

Tried 2 methods for each of these...
1) minidsp app on Win10 connected directly to MiniDSP 2x4HD via USB
2) minidsp app on Linux using remote (--tcp <ip:5333>) to minidsp app on Win10 (#1 above)

Attempted following commands:
minidsp input 0 peq all clear
minidsp input 0 peq all bypass on (or off)
minidsp input 0 peq all import <file>
minidsp input 1 peq all clear
minidsp input 1 peq all bypass on (or off)
minidsp input 1 peq all import <file>

The server side shows the config (example here, for bypass):
Sent: WriteBiquadBypass { addr: 20b7, value: true }
Recv: Ack
Recv: Ack
Sent: WriteBiquadBypass { addr: 20b2, value: true }
Recv: Ack
Recv: Ack
Sent: WriteBiquadBypass { addr: 20ad, value: true }
Recv: Ack
Recv: Ack
Sent: WriteBiquadBypass { addr: 20a8, value: true }
Recv: Ack
Recv: Ack
Sent: WriteBiquadBypass { addr: 20a3, value: true }
Recv: Ack
Sent: WriteBiquadBypass { addr: 209e, value: true }
Recv: Ack
Recv: Ack
Recv: Ack
Sent: WriteBiquadBypass { addr: 2099, value: true }
Recv: Ack
Recv: Ack
Sent: WriteBiquadBypass { addr: 2094, value: true }
Recv: Ack
Recv: Ack
Sent: WriteBiquadBypass { addr: 208f, value: true }
Recv: Recv: AckAck

Sent: WriteBiquadBypass { addr: 208a, value: true }
Recv: Ack
Recv: Ack

mrene, 2021-01-11:
Thanks for the report.

Do you mean that you can't hear the difference, or that their app isn't reporing it? (It's normal that the minidsp application doesn't report changes, because the device doesn't support reading configuration back from the device)

Can you try the following? It works fine on my side on a 2x4HD. If it doesn't, we can do more steps to figure out what the app does different than the plugin by inspecting the output of `server`

Use the following filter - it's a high shelf around 200hz, it'll be really obvious if it's applied.

Save this as `filt.txt`

```txt
biquad1,
b0=0.25296561425934055,
b1=-0.4987820769827858,
b2=0.24591409251506618,
a1=1.9858327346088447,
a2=-0.9859303644004657
```

```shell
minidsp input 1 mute on
minidsp input 0 peq all import ./filt.txt
minidsp input 0 peq all bypass off

# Listen to lowpassed audio on chan0

minidsp input 1 mute off
minidsp input 0 mute on
minidsp input 0 peq all import ./filt.txt
minidsp input 0 peq all bypass off

# Listen to lowpassed audio on chan1

# Bypass everything back
minidsp input 0 peq all bypass on
minidsp input 1 peq all bypass on
```

linerjoe, 2021-01-11:
The MiniDSP app shows the change of config from 1 to 2, etc. Is that the only change it recognizes?

In other words, there isn’t a way to actually verify what the current config is on the MiniDSP?

I will try your test tomorrow, regarding add a filter to verify. I obviously assumed (incorrectly), that the app would show the changes.

Joe

mrene, 2021-01-11:
The master state can be read, that includes:
- master gain and mute
- current configuration preset
- active input source
- a random value to detect configuration desync 

All the other settings are read from a local XML file. In the future I might add a way to keep it in sync so it's always possible to look at the config.

Changing the filters to obviously noticable ones is how I have been testing things so far.

mrene, 2021-01-12:
Closing this since it seems to work fine as per your avsforum post, feel free to re-open if you run into problems

closed 2021-01-12 by mrene