mrene/minidsp-rs #21: Intermittent "Error: A malformed packet was received"

conrad-nash, 2021-01-12:
I have minidsp running in server mode on an rpi connected via USB to the minidsp 2x4.
I'm able to communicate with the minidsp over tcp from a windows 10 machine using: `minidsp --tcp 192.168.86.40:5333`
Server is 192.168.86.40, client is 192.168.86.54.

When attempting to clear or bypass all filters for an input I get the message: Error: A malformed packet was received

This is the command I'm sending: `minidsp --tcp 192.168.86.40:5333 input 1 peq all bypass on`

If I use `minidsp --tcp 192.168.86.40:5333 input 1 peq all clear` I get an error every time.

Here's the log from the server

Working call:

`New connection: 192.168.86.54:50768
Sent: WriteBiquadBypass { addr: 20b7, value: true } (0x20b7 | 8375) <> PEQ_2_10
Recv: Ack
Recv: Ack
Sent: WriteBiquadBypass { addr: 20b2, value: true } (0x20b2 | 8370) <> PEQ_2_9
Recv: Ack
Recv: Ack
Sent: WriteBiquadBypass { addr: 20ad, value: true } (0x20ad | 8365) <> PEQ_2_8
Recv: Ack
Recv: Ack
Sent: WriteBiquadBypass { addr: 20a8, value: true } (0x20a8 | 8360) <> PEQ_2_7
Recv: Ack
Recv: Ack
Sent: WriteBiquadBypass { addr: 20a3, value: true } (0x20a3 | 8355) <> PEQ_2_6
Recv: Ack
Recv: Ack
Sent: WriteBiquadBypass { addr: 209e, value: true } (0x209e | 8350) <> PEQ_2_5
Recv: Ack
Recv: Ack
Sent: WriteBiquadBypass { addr: 2099, value: true } (0x2099 | 8345) <> PEQ_2_4
Recv: Ack
Recv: Ack
Sent: WriteBiquadBypass { addr: 2094, value: true } (0x2094 | 8340) <> PEQ_2_3
Recv: Ack
Recv: Ack
Sent: WriteBiquadBypass { addr: 208f, value: true } (0x208f | 8335) <> PEQ_2_2
Recv: Ack
Recv: Ack
Sent: WriteBiquadBypass { addr: 208a, value: true } (0x208a | 8330) <> PEQ_2_1
Recv: Ack
Recv: Ack
Closed: 192.168.86.54:50768`

Response that resulted in an error on the client:

`New connection: 192.168.86.54:50772
Sent: WriteBiquadBypass { addr: 20b7, value: true } (0x20b7 | 8375) <> PEQ_2_10
Recv: Ack
Recv: Ack
Sent: WriteBiquadBypass { addr: 20b2, value: true } (0x20b2 | 8370) <> PEQ_2_9
Recv: Recv: Ack
Ack
Sent: WriteBiquadBypass { addr: 20ad, value: true } (0x20ad | 8365) <> PEQ_2_8
Recv: Ack
Recv: Ack
Closed: 192.168.86.54:50772
`
Also, two issues in your readme.
In the input PEQ section the command examples have output listed instead of input, and in both input and output sections your last peq example says preq instead of peq.

mrene, 2021-01-12:
Do you happen to have the minidsp plugin app open and connected at the same time? I have a slight idea of what can cause this, I'll push something to try out a bit later.

mrene, 2021-01-13:
I added more specific detail if that error would be to occur again. Can you give [0.0.3](https://github.com/mrene/minidsp-rs/releases/tag/v0.0.3) a try?

conrad-nash, 2021-01-13:
I tried this tonight and I can't replicate the error with 0.0.3 running as both the client and the server.
Even with the official plug-in connected at the same time firing multiple requests to clear and bypass all always results in a positive outcome - nice job!

closed 2021-01-13 by conrad-nash