mrene/minidsp-rs #25: Cargo install fails on Pi3

conrad-nash, 2021-01-13:
After installing Cargo using apt-get on a fresh install of Raspbian OS, I'm unable to complete an install of minidsp using `cargo install minidsp`

Here's my output:

````root@raspberrypi:/home/pi/Downloads# cargo install minidsp
    Updating crates.io index
  Downloaded minidsp v0.0.3
  Downloaded 1 crate (67.5 KB) in 1.89s
  Installing minidsp v0.0.3
  Downloaded futures-sink v0.3.10
  Downloaded futures-io v0.3.10
  Downloaded futures v0.3.10
  Downloaded tokio-stream v0.1.2
  Downloaded futures-util v0.3.10
  Downloaded futures-core v0.3.10
  Downloaded futures-executor v0.3.10
  Downloaded futures-channel v0.3.10
  Downloaded futures-task v0.3.10
  Downloaded futures-macro v0.3.10
   Compiling proc-macro2 v1.0.24
   Compiling unicode-xid v0.2.1
   Compiling syn v1.0.58
   Compiling libc v0.2.82
   Compiling autocfg v1.0.1
   Compiling memchr v2.3.4
   Compiling cfg-if v1.0.0
   Compiling version_check v0.9.2
   Compiling log v0.4.13
   Compiling futures-core v0.3.10
   Compiling once_cell v1.5.2
   Compiling proc-macro-hack v0.5.19
   Compiling pin-project-lite v0.2.4
   Compiling lazy_static v1.4.0
   Compiling cfg-if v0.1.10
   Compiling futures-sink v0.3.10
   Compiling proc-macro-nested v0.1.6
   Compiling scopeguard v1.1.0
   Compiling smallvec v1.6.1
   Compiling futures-io v0.3.10
   Compiling adler v0.2.3
   Compiling gimli v0.23.0
   Compiling slab v0.4.2
   Compiling bytes v1.0.1
   Compiling rustc-demangle v0.1.18
   Compiling jetscii v0.4.4
   Compiling regex-syntax v0.6.22
   Compiling unicode-segmentation v1.7.1
   Compiling object v0.22.0
   Compiling cc v1.0.66
   Compiling bitflags v1.2.1
   Compiling pkg-config v0.3.19
   Compiling pin-utils v0.1.0
   Compiling unicode-width v0.1.8
   Compiling termcolor v1.1.2
   Compiling anyhow v1.0.38
   Compiling hashbrown v0.9.1
   Compiling vec_map v0.8.2
   Compiling humantime v2.0.1
   Compiling termcolor v0.3.6
   Compiling strsim v0.10.0
   Compiling xmlparser v0.13.3
   Compiling riff v1.0.1
error[E0599]: no associated item named `MAX` found for type `u32` in the current scope
  --> /root/.cargo/registry/src/github.com-1ecc6299db9ec823/riff-1.0.1/src/lib.rs:97:37
   |
97 |         if data.len() as u64 > u32::MAX as u64 {
   |                                     ^^^ associated item not found in `u32`
   |
help: you are looking for the module in `std`, not the primitive type
   |
97 |         if data.len() as u64 > std::u32::MAX as u64 {
   |                                ^^^^^^^^^^^^^

error[E0599]: no associated item named `MAX` found for type `u32` in the current scope
   --> /root/.cargo/registry/src/github.com-1ecc6299db9ec823/riff-1.0.1/src/lib.rs:123:29
    |
123 |         if total_len > u32::MAX as u64 {
    |                             ^^^ associated item not found in `u32`
    |
help: you are looking for the module in `std`, not the primitive type
    |
123 |         if total_len > std::u32::MAX as u64 {
    |                        ^^^^^^^^^^^^^

error[E0599]: no associated item named `MAX` found for type `u32` in the current scope
   --> /root/.cargo/registry/src/github.com-1ecc6299db9ec823/riff-1.0.1/src/lib.rs:145:29
    |
145 |         if total_len > u32::MAX as u64 {
    |                             ^^^ associated item not found in `u32`
    |
help: you are looking for the module in `std`, not the primitive type
    |
145 |         if total_len > std::u32::MAX as u64 {
    |                        ^^^^^^^^^^^^^

error: aborting due to 3 previous errors

For more information about this error, try `rustc --explain E0599`.
error: could not compile `riff`.
warning: build failed, waiting for other jobs to finish...
error: failed to compile `minidsp v0.0.3`, intermediate artifacts can be found at `/tmp/cargo-installbAOOuo`

Caused by:
  build failed
````
I have that referenced folder if that's of any use?

mrene, 2021-01-13:
My hunch would be that you have an older rust version, you could try installing from [rustup](https://rustup.rs/)

Btw there is a package for raspbian that is automaticall built: 
https://github.com/mrene/minidsp-rs/releases/download/v0.0.3/minidsp_0.0.3_armhf.deb

conrad-nash, 2021-01-13:
Installing rustup and then attempting to install got further but then still failed.
If you want me to continue investigating I'm happy to. I was able to get the service working by installing the deb package manually though.

Let me know if you want to carry on with cargo otherwise I'm happy to close this.

mrene, 2021-01-13:
You need some development packages installed to be able to link with libusb, `sudo apt-get install libusb-1.0-0 libusb-1.0-0-dev` - but if you don't plan on changing the code you're better off with the packages

closed 2021-01-13 by conrad-nash