mrene/minidsp-rs #39: Provide machine readable status output

3ll3d00d, 2021-01-17:
current output is easy to read but fiddly to parse, an option to format this output in some way would be useful e.g.

`minidsp -o json`

would provide the same output but as a json document

mrene, 2021-01-18:
Are you looking at other things except master status?

I have this in progress:

```json
{
  "master": {
    "preset": 0,
    "source": "Toslink",
    "volume": -25.0,
    "mute": false
  },
  "available_sources": [
    "Analog",
    "Toslink",
    "Usb"
  ],
  "input_levels": [
    -131.42198,
    -131.42198
  ],
  "output_levels": [
    -157.00223,
    -166.65083,
    -120.0,
    -120.0
  ]
}
```

3ll3d00d, 2021-01-18:
atm the only value I read is  `master.preset`

closed 2021-01-18 by mrene

mrene, 2021-01-19:
0.0.5 has `-o json` which pretty prints the above json, or `-o jsonline` which prints it on a single line, which should be easier to parse.

If you're keeping an open subprocess, I added a `status` command, just run it as `minidsp -o jsonline -f -` and it'll echo back the json object.

I have to look at this http interface next :)

PR #40 (2021-01-18): feat(cli): add json output to master status cmd
Closes #39