mrene/minidsp-rs #48: Wi-DG not detected by `minidsp probe`

mrene, 2021-01-24:


mrene, 2021-01-28:
Got some reports that the WI-DG broadcasts a udp frame every ~10s or so, `probe` only waits for 2s.

mrene, 2021-02-06:
WIll be addressed by #58 since the daemon can keep listening for these packets and update its list.

closed 2021-02-06 by mrene

langmeister, 2022-03-11:
Can you provide a little more info on how the discovery works or prerequisites/dependencies or how to debug the probe feature?

I have 0.1.4 installed in a very basic debian docker container configured to run on same net as host.  Connecting to the Wi-DG when specifying IP (--tcp=w.x.y.z) works perfectly, but 'minidsp probe' just returns the command prompt right away.

mrene, 2022-03-11:
You need `minidsp probe --net` if you want to wait for wi-dg broadcasts.

It basically listens for broadcast packets on udp port 3999 for a few seconds until it receives one. It should work in docker with `--net host`.

If you run minidspd, it'll maintain a list of discovered local devices and allow you to talk to it directly. The command line app looks for a unix socket at `/tmp/minidsp.sock` and uses it to find available devices through the local minidspd instance.

langmeister, 2022-03-12:
Thank you.  This greatly helped with troubleshooting.

Issue was my NAS was connected to both networks, and the docker container was bound to my trusted LAN interface instead of the one for my IoT subnet where the MiniDSP resides, and the UDP packets were not being passed across the gateway. Putting them both on the same subnet did the trick and all works perfectly now.