mrene/minidsp-rs #56: Failure:  Transport has closed

MervisDiamonds, 2021-02-02:
I'm running the minidsp-rs server and I received the following error after starting the service.  Is this something I need to address, if so how?

`Feb 02 05:22:20 raspberrypi systemd[1]: Started minidsp.
Feb 02 06:00:04 raspberrypi minidsp[978]: [2021-02-02T11:00:04Z ERROR minidsp::server] stream rx: Err(Lagged(1))
Feb 02 06:00:04 raspberrypi minidsp[978]: Error: Transport has closed
Feb 02 06:00:04 raspberrypi systemd[1]: minidsp.service: Main process exited, code=exited, status=1/FAILURE
Feb 02 06:00:04 raspberrypi systemd[1]: minidsp.service: Failed with result 'exit-code'.`

mrene, 2021-02-02:
Does the service restart afterwards? You definitely seem to have hit a bug but it looks like something that wouldn't happen every time (and the service is configured to auto-restart)

MervisDiamonds, 2021-02-02:
It did a couple of times last night and this morning but seems to be ok today.

mrene, 2021-06-05:
This is largely fixed in the current 0.1.0 pre-releases.

closed 2021-06-05 by mrene