mrene/minidsp-rs #91: feature request: list of known working devices please?

zogith, 2021-04-05:
Hi,

Can you add a list of MiniDSP devices this is known to work with in the README
Does it work with the old "classic" miniDSP 2x4?

kind regards.

mrene, 2021-04-05:
Hi there,

I haven't heard much feedback about other devices yet. The only list I could infer from the wire protocol is [the one from the WI-DG](https://www.minidsp.com/products/accessories/wi-dg).

For these devices, some features don't require special support (master volume/mute, source selection, preset selection), but  anything else does. 

I've been asking people that have other hardware to upload:
- the output of lsusb
- the output of `minidsp debug id`
- an example xml configuration file form the official app

With these I can sort of guess the meaning of each parameter.

This being said, I'd be happy to add any reports of it working or not working with other devices, let me know how it goes with the classic 2x4.

zogith, 2021-04-05:
ok thanks.. when I get a Linux laptop with USB close to my miniDSP 2x4 I'll give it a bash, I'm curious :-) 

but, the documentation doesn't mention (except for a hint hidden away) what miniDSP devices this is known to work on. Does it work on the miniDSP 2x4 HD?

mrene, 2021-04-05:
Yes it does, the 2x4HD is the only device that is 100% supported at the moment. I'll make this clearer :)

zogith, 2021-04-05:
great, I found this while prepping for an upgrade from miniDSP 2x4 to miniDSP 2x4 HD and linux support is yet another reason to upgrade :-)
thanks for your great work, open source FTW!

zogith, 2021-04-14:
Hi,

my miniDSP 2x4 HD has arrived so I will move over to it soon, but while I'm still poking around with the miniDSP 2x4 "classic", here is the info you requested when it is plugged in...

output of lsusb: (only the new line after plugging it in):

Bus 001 Device 011: ID 04d8:003f Microchip Technology, Inc. 

when plugged in dmesg says:

[31877.464669] usb 1-11.1: New USB device found, idVendor=04d8, idProduct=003f, bcdDevice= 0.02
[31877.464675] usb 1-11.1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[31877.464678] usb 1-11.1: Product: MiniDSP v1.2
[31877.464681] usb 1-11.1: Manufacturer: DSP4YOU LTD.
[31877.475608] hid-generic 0003:04D8:003F.0009: hiddev4,hidraw5: USB HID v1.11 Device [DSP4YOU LTD. MiniDSP v1.2] on usb-0000:00:14.0-11.1/input0

the output of "minidsp debug id": (and this works OK for the miniDSP 2x4 HD)..

minidsp debug id
Error: Couldn't find any MiniDSP devices

An example XML plugin file, renamed as TXT as github didnt like xml extension

[summer20152016-tda7297-and-irs2092-lowerhf.txt](https://github.com/mrene/minidsp-rs/files/6311168/summer20152016-tda7297-and-irs2092-lowerhf.txt)

mrene, 2021-04-14:
Can you try:

```
minidsp --usb 04d8:003f debug id
```

zogith, 2021-04-15:
> Can you try:
> 
> ```
> minidsp --usb 04d8:003f debug id
> ```

minidsp --usb 04d8:003f debug id
Probing local hid devices:
No matching local USB devices detected.

thread 'tokio-runtime-worker' panicked at 'range start must not be greater than end: 1 <= 0', /home/runner/.cargo/registry/src/github.com-1ecc6299db9ec823/bytes-1.0.1/src/bytes.rs:245:9
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace

is the result

mrene, 2021-04-15:
Hmm that looks like it can't see the device at all

mrene, 2021-05-18:
Device list added in #122

closed 2021-05-18 by mrene