mrene/minidsp-rs #115: Support for asynchronous reading of minidsp status

kukocz, 2021-05-11:
Hi,

As we discussed [HERE in "Discussions" section](https://github.com/mrene/minidsp-rs/discussions/110#discussioncomment-701563) it would be a good idea to implement asynchronous device events (to get updates on the same way as native desktop app does). It could be used i.e. to get regular updates on OLED display, instead of sending continuously commands to get the status of the device (needs a lot of cpu power).

[image]

mrene, 2021-05-13:
I added a way to get updates via the HTTP API in #112. There is a caveat, these updates have to be made from the IR remote, other ways to update settings (using the plugin app, or minidsp commands) don't trigger messages (except maybe when the config preset changes. 

Does that work for you?

kukocz, 2021-05-14:
Does it mean, that only gain meter values are constantly updated?
Nonetheless it should solve my problem (remote triggered events should be updating display instantly). Rotary encoder based changes will trigger minidsp commands separately.
Thank you and can't wait to download updated pre-release version.

mrene, 2021-05-15:
You will get updates for these:
- master mute
- master volume
- active configuration preset
- active source

I strongly suggest you do updates [via the API](https://minidsp-rs.mrene.workers.dev/#/paths/~1devices~1{deviceIndex}~1config/post) instead of forking a call to `minidsp`, you'll save a lot on cpu+time since everything is already loaded and ready to use.

For example, this toggles the global mute status:
```
curl -d '{ "master_status": { "mute": true } }' http://minidsp.local:5380/devices/0/config
```

(1.0.1-pre2 contains these changes)

mrene, 2021-05-15:
almost forgot, dirac live status shoudl also be part of these updates

kukocz, 2021-05-17:
Hi,
I'm testing v0.1.0-pre2 since yesterday and websocket/api control is really a step forward! I'm already using websockets for continuous monitor of updates. I'm about to use it for setting values via api too.

Issues I noticed so far:
- remote control doesn't trigger events for source changes
- DDRC-24 is not correctly recognized (although Dirac control is working well):
```
[
  {
    "url": "usb:0001%3A0003%3A04?vid=2752&pid=0044",
    "version": {
      "hw_id": 10,
      "dsp_version": 101,
      "serial": 965535
    },
    "product_name": "2x4HD"
  }
]
```
```
$ minidsp probe
Found 2x4HD with serial 965535 at ws://localhost/devices/0/ws [hw_id: 10, dsp_version: 101]
```
- can't read input/output levels (always zero's)
```
{
  "master": {
    "preset": 1,
    "source": "Toslink",
    "volume": -37.0,
    "mute": false,
    "dirac": true
  },
  "input_levels": [
    0.0,
    0.0
  ],
  "output_levels": [
    0.0,
    0.0,
    0.0,
    0.0
  ]
}
```
```
 $ minidsp
MasterStatus { preset: 1, source: Toslink, volume: Gain(-37.0), mute: false, dirac: true }
Input levels: 0.0, 0.0
Output levels: 0.0, 0.0, 0.0, 0.0
```

mrene, 2021-05-17:
Hmm I think I wrongly assumed the 2x4HD and the DDRC would have the same address layout. Could you attach your XML config? The actual settings don't matter, but the file contains all the addresses needed to talk to the device.

`minidsp debug id` should also display something like this:

```
Dumping readable floats:
0044: -131.39944
0045: -131.39944
004a: -140.30893
004b: -139.93767
004c: -120.0
004d: -120.0
```

Could you paste what yours is printing? The other conversartion was truncated before this output.

mrene, 2021-05-17:
I can reproduce the source issue, they use a different address for the async event than they do for the normal status fetch. 

Can you enable/disable dirac live from the remote? Is that reported properly?

kukocz, 2021-05-18:
Hi,

Dirac enable/disable via remote is working correctly (triggers update events).
When it comes to the full debug listening, here are the results:

Current status:
```
$ minidsp
MasterStatus { preset: 1, source: Toslink, volume: Gain(-41.0), mute: false, dirac: true }
Input levels: 0.0, 0.0
Output levels: 0.0, 0.0, 0.0, 0.0
```
debug id:

```
$ minidsp debug id
Probing local hid devices:
No matching local USB devices detected.

HW ID: 10
DSP Version: 101
Detected sources: [(Analog, 0), (Toslink, 1), (Usb, 2)]

Dumping memory:
len=95
0000FFA0  FF 65 FF 03 20 03 2C 01 03 00 64 00 00 00 00 00  | .e.♥ ♥,☺♥.d..... |
0000FFB0  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  | ................ |
0000FFC0  00 00 FF FF F8 FF FF FF 05 DC A6 77 01 98 53 39  | ....▫...♣▄ªw☺ÿS9 |
0000FFD0  01 81 5F 42 02 10 3A 3B 01 01 52 00 FF FF FF FF  | ☺ü_B☻►:;☺☺R..... |
0000FFE0  00 00 00 00 00 00 FF FF FF FF FF FF FF FF FF FF  | ................ |
0000FFF0  FF FF FF FF FF FF FF FF FF FF FF FF FF FF 06     | ..............♠  |

Dumping readable floats:
0008: -10.3229065
000d: -10.3229065
0010: -120.0
0011: -120.0
0012: -120.0
0013: -120.0
0014: -120.0
0015: -120.0
0016: -120.0
0017: -120.0
0018: -120.0
0019: -120.0
001a: -120.0
001b: -120.0
001c: -120.0
001d: -120.0
001e: -120.0
001f: -120.0
0020: 0.000000000000000000000000000000000000000000003
0021: 0.000000000000000000000000000000000000000000001
0023: -30.0
002a: -24.062094
002c: -66.258286
0060: 0.0038172458
0061: 0.0076344917
0062: 0.0038172458
0063: 0.0076344917
0064: 0.0038172458
0065: 1.7695043
0066: -0.78477335
0067: 0.0038172458
0068: 0.0076344917
0069: 0.0038172458
006a: 1.7695043
006b: -0.78477335
006c: 0.0040740687
006d: 0.0081481375
006e: 0.0040740687
006f: 1.888556
0070: 0.0040740687
0071: 0.0081481375
0072: 0.0040740687
0073: 1.888556
0074: -0.9048522
0075: 0.8885694
0076: -1.7771388
0077: 0.8885694
0078: 1.7695043
0079: -0.78477335
007a: 0.8885694
007b: -1.7771388
007c: 0.8885694
007d: 1.7695043
007e: 0.94835204
007f: -1.8967041
0080: 0.94835204
0081: 1.888556
0082: -0.9048522
0083: 0.94835204
0084: -1.8967041
0085: 0.94835204
0086: 1.888556
0087: -0.9048522
0088: 0.000000000000000000000000000000000000000000003
0089: 0.000000000000000000000000000000000000000000001
008b: -30.0
0092: -21.3014
0094: -62.289444
00c8: 0.06613022
00c9: 0.13226044
00ca: 0.06613022
00cb: 1.0155429
00cc: -0.28006372
00cd: 0.06613022
00ce: 0.13226044
00cf: 0.06613022
00d0: 1.0155429
00d1: -0.28006372
00d2: 0.083800845
00d3: 0.16760169
00d4: 0.083800845
00d5: 1.2869054
00d6: -0.6221088
00d7: 0.083800845
00d8: 0.16760169
00d9: 0.083800845
00da: 1.2869054
00db: -0.6221088
00dc: 0.8885694
00dd: -1.7771388
00de: 0.8885694
00df: 1.7695043
00e0: 0.8885694
00e1: -1.7771388
00e2: 0.8885694
00e3: 1.7695043
00e4: -0.78477335
00e5: 0.8885694
00e6: -1.7771388
00e7: 0.8885694
00e8: 1.7695043
00e9: -0.78477335
00ea: 0.94835204
00eb: -1.8967041
00ec: 0.94835204
00ed: 1.888556
00ee: 0.94835204
00ef: -1.8967041
00f0: 0.94835204
00f1: 1.888556
00f2: -0.9048522
00f3: -30.0
00fa: -120.0
00fb: NaN
00fc: -120.0
```

mrene, 2021-05-18:
Oh yeah that's a very different output than on the 2x4hd. Once you attach your XML config I'll be able to hook it up so it uses the right addresses.

kukocz, 2021-05-18:
Hi,

Please find attached config file. I hope it will help :)
[config.xml.txt](https://github.com/mrene/minidsp-rs/files/6501822/config.xml.txt)

mrene, 2021-06-05:
Got some time to look into this, I mapped the device addresses based on what I think it should be, although I may have made mistake as to what levels maps to what because there were multiple config params that seemed to fit each purpose.

Now that the device is going to be recognized as a ddrc24, it's possible to use the plugin app and figure out what these mappings could be, there is a description of the procedure here: https://github.com/mrene/minidsp-rs/pull/105#issuecomment-828915750 

You could confirm the meter levels are okay to start with (you'll need the final 0.1.0 build, or to build from source)

closed 2021-06-05 by mrene