mrene/minidsp-rs #134: device status is unclear in 0.1

3ll3d00d, 2021-06-04:
I copied this over from the 0.1 thread. I'm not sure what the root cause is but it looks like the http or ws api can be pushed into a fundamentally broken state

At some point, I hit /devices and get

```
  {
    "url": "usb:0001%3A003f%3A04?vid=2752&pid=0011",
    "version": null,
    "product_name": null
  }
]
```

it's not obvious whether this is a good or bad thing

/devices/0 says

```
{
  "message": "the specified device is not ready to accept requests",
  "error": {
    "type": "DeviceNotReady"
  }
}
```

this also returns a 500 status code

if I hit the websocket then this also returns a 500 which is (I think) an unusual response for a websocket

however if I use the cli then I get

```
note that the cli at this point responds normally

```
$ minidsp                                                                                                                                                                               
MasterStatus { preset: 1, source: Usb, volume: Gain(-2.0), mute: false, dirac: false }
Input levels: -131.4, -131.4
Output levels: -125.3, -125.3, -120.0, -125.3
```

i.e. an apparently functioning system

after a `sudo systemctl restart minidsp` then I get normal responses and the ws is happy again

3ll3d00d, 2021-06-04:
simple steps to reproduce

1) connect device via usb
2) run cli and http or ws commands to verify device is ok
3) unplug usb cable
4) run `minidsp` (results in `No devices found`)
5) plug usb cable in
6) run `minidsp` (get normal status output)
7) hit rest api at devices/1 and get a 500 wit

```
{
  "message": "the specified device is not ready to accept requests",
  "error": {
    "type": "DeviceNotReady"
  }
```

8) run `sudo systemctl restart minidsp`
9) hit /devices/0 again and get a 200 with the expected (normal) output showing healthy device

i.e. web server api looks like it is blowing up on device disconnection

closed 2021-06-05 by mrene

mrene, 2021-06-05:
Should be all fixed in [pre3](https://github.com/mrene/minidsp-rs/releases/tag/v0.1.0-pre3)

PR #136 (2021-06-05): fix(daemon): refresh usb device list at every iteration
- fix(usb): actually refresh the device list at each discovery cycle
- bump rpi rust version

Fixes #134