mrene/minidsp-rs #135: ws only updates when cli executes

3ll3d00d, 2021-06-04:
I have a client listening to the websocket on `ws://localhost/devices/0?levels=true` via `/tmp/minidsp.sock` and a separate shell open 

execution of `minidsp` in the shell immediately produces `{"master":{"preset":1,"source":"Usb","volume":-4.0,"mute":false}}` on the websocket

if I then do

```
$ minidsp -v gain -- -5                                                                                                                                                                              
 
Sent: ReadHardwareId 
Recv: HardwareId { payload: [01, 09, 0a] }
Sent: ReadMemory { addr: ffa1, size: 01 } 
Recv: MemoryData(MemoryView { base: ffa1, data: [64] })
Sent: ReadMemory { addr: fffc, size: 02 } 
Recv: MemoryData(MemoryView { base: fffc, data: [ff, ff] })
Sent: SetVolume { value: Gain(-5.0) } 
Recv: Ack

```

nothing is received via the ws

then

```
$ minidsp
MasterStatus { preset: 1, source: Usb, volume: Gain(-5.0), mute: false, dirac: false }
Input levels: -131.4, -131.4
Output levels: -128.4, -128.4, -120.0, -128.4
```

shows the command was accepted and gain updated

and the ws receives

```
{"master":{"preset":1,"source":"Usb","volume":-5.0,"mute":false}}
```

so it seems like the ws is not updating as expected

mrene, 2021-06-05:
Async status updates through websocket currently only reflect operations that were done through the device's IR remote, which. due to some protocol similarities, are also triggered by master status requests, causing this weird looking behaviour. 

There will eventually be a way to get updates done by other minidsp-rs clients (including the official plugin app), but right now the command line tool bypasses the API and directly sends raw frames to the device, making it non-trivial to broadcast update operations to other clients.

mrene, 2021-06-05:
This being said I can probably poll the device for changes once in a while, won't be truly async so I'd control it via another query string param

mrene, 2021-06-05:
Polling is now available in [0.1.0](https://github.com/mrene/minidsp-rs/releases/tag/v0.1.0)

closed 2021-06-05 by mrene

mrene, 2021-06-05:
The [docs](https://minidsp-rs.pages.dev/daemon/http#websocket-streaming) have been updated to, if you pass `poll=true, you'll get updates whenever the status has changed, which gets checked every 2s.