mrene/minidsp-rs #142: DDRC-24 is not correctly recognized - v0.1.0

kukocz, 2021-06-05:
Hi,

Just updated minidsp-re to the newest version. I can confirm that Source change via remote triggers update events now but device is still recognized as 2x4HD instead of DDRC-24;

```
$ minidsp probe
Found 2x4HD with serial 965535 at usb:0001%3A0004%3A04?vid=2752&pid=0044 [hw_id: 10, dsp_version: 101]
```
Readings in verbose mode (connected via official plugin app):
```
$ minidsp -v server
Sent: ReadHardwareId
Recv: HardwareId { payload: [01, 0b, 0a] }
Sent: ReadMemory { addr: ffa1, size: 01 }
Recv: MemoryData(MemoryView { base: ffa1, data: [65] })
Sent: ReadMemory { addr: fffc, size: 02 }
Recv: MemoryData(MemoryView { base: fffc, data: [ff, ff] })
Sent: ReadMemory { addr: ffa1, size: 01 }
Recv: MemoryData(MemoryView { base: ffa1, data: [65] })
Sent: ReadHardwareId
Recv: HardwareId { payload: [01, 0b, 0a] }
Sent: ReadMemory { addr: ffc8, size: 10 }
Recv: MemoryData(MemoryView { base: ffc8, data: [05, dc, a6, 77, 01, 98, 53, 39, 01, 81, 5f, 42, 02, 10, 3a, 3b] })
Sent: ReadMemory { addr: ffd9, size: 01 }
Recv: MemoryData(MemoryView { base: ffd9, data: [01] })
Sent: ReadMemory { addr: ffd8, size: 01 }
Recv: MemoryData(MemoryView { base: ffd8, data: [01] })
Sent: ReadMemory { addr: fffe, size: 02 }
Recv: MemoryData(MemoryView { base: fffe, data: [06, cb] })
Sent: Unknown { cmd_id: 40, payload: [] }
Recv: Unknown { cmd_id: 40, payload: [40, 15, 24] }
Sent: ReadMemory { addr: ffe5, size: 01 }
Recv: MemoryData(MemoryView { base: ffe5, data: [00] })
Sent: ReadMemory { addr: ffe0, size: 01 }
Recv: MemoryData(MemoryView { base: ffe0, data: [00] })
Sent: ReadMemory { addr: ffda, size: 02 }
Recv: MemoryData(MemoryView { base: ffda, data: [41, 00] })
Sent: Unknown { cmd_id: 45, payload: [a1, 03, 00, 00, 02] }
Recv: Unknown { cmd_id: 45, payload: [45, a1, 03, 00, 03, ff, b8, 00, 00, 00, 00] }
Sent: SetMute { value: true }
Recv: Ack
Sent: WriteMemory { addr: ffcc, data: [02, ac, db, 37] }
Recv: Ack
Sent: SetMute { value: false }
Recv: Ack
```
Because of the above the input levers are all zeros:
```
$ minidsp
MasterStatus { preset: 1, source: Toslink, volume: Gain(-32.5), mute: false, dirac: true }
Input levels: 0.0, 0.0
Output levels: 0.0, 0.0, 0.0, 0.0
```

mrene, 2021-06-05:
It looks like I forgot to add it to the devices module, time for 0.1.1 :D

mrene, 2021-06-05:
Do you have any idea what this corresponds to? Was it triggered by an action?

```
Sent: Unknown { cmd_id: 45, payload: [a1, 03, 00, 00, 02] }
Recv: Unknown { cmd_id: 45, payload: [45, a1, 03, 00, 03, ff, b8, 00, 00, 00, 00] }
```

mrene, 2021-06-05:
Fix [released](https://github.com/mrene/minidsp-rs/releases/tag/v0.1.1)

kukocz, 2021-06-05:
> It looks like I forgot to add it to the devices module, time for 0.1.1 :D

Shit happens :D I'm waiting patiently for the fix

> Do you have any idea what this corresponds to? Was it triggered by an action?
> 
> ```
> Sent: Unknown { cmd_id: 45, payload: [a1, 03, 00, 00, 02] }
> Recv: Unknown { cmd_id: 45, payload: [45, a1, 03, 00, 03, ff, b8, 00, 00, 00, 00] }
> ```

I double checked. It's the response received when official plugin connected. No other command/action sent.

kukocz, 2021-06-05:
Good and bad news :)

DDRC-24 is correctly recognized:
```
$ minidsp probe
Found DDRC-24 with serial 965535 at ws://localhost/devices/0/ws [hw_id: 10, dsp_version: 101]
```
Levels are read but not correct:
```
$ minidsp
MasterStatus { preset: 1, source: Toslink, volume: Gain(-48.0), mute: false, dirac: true }
Input levels: -120.0, -120.0
Output levels: -120.0, -120.0, -120.0
```

mrene, 2021-06-05:
Ahhh, can you run `minidsp server -vvv`? It should print repeated lines when the plugin app is fetching input levels. They will differ for input and outputs, so make sure to switch to the output section so both are reported.

kukocz, 2021-06-05:
Here's the output:
(readings are continuous - I changed to output tab after w while then terminated server after a 1-2 seconds)

```
$ minidsp -vvv server
Sent: [31]
Sent: ReadHardwareId
Recv: [31, 01, 0b, 0a]
Recv: HardwareId { payload: [01, 0b, 0a] }
Sent: [05, ff, a1, 01]
Sent: ReadMemory { addr: ffa1, size: 01 }
Recv: [05, ff, a1, 65]
Recv: MemoryData(MemoryView { base: ffa1, data: [65] })
Sent: [05, ff, fc, 02]
Sent: ReadMemory { addr: fffc, size: 02 }
Recv: [05, ff, fc, ff, ff]
Recv: MemoryData(MemoryView { base: fffc, data: [ff, ff] })
Sent: [05, ff, a1, 01]
Sent: ReadMemory { addr: ffa1, size: 01 }
Recv: [05, ff, a1, 65]
Recv: MemoryData(MemoryView { base: ffa1, data: [65] })
Sent: [31, 01]
Sent: ReadHardwareId
Recv: [31, 01, 0b, 0a]
Recv: HardwareId { payload: [01, 0b, 0a] }
Sent: [05, ff, c8, 10]
Sent: ReadMemory { addr: ffc8, size: 10 }
Recv: [05, ff, c8, 05, dc, a6, 77, 02, ac, db, 37, 01, 81, 5f, 42, 02, 10, 3a, 3b]
Recv: MemoryData(MemoryView { base: ffc8, data: [05, dc, a6, 77, 02, ac, db, 37, 01, 81, 5f, 42, 02, 10, 3a, 3b] })
Sent: [05, ff, d9, 01]
Sent: ReadMemory { addr: ffd9, size: 01 }
Recv: [05, ff, d9, 02]
Recv: MemoryData(MemoryView { base: ffd9, data: [02] })
Sent: [05, ff, d8, 01]
Sent: ReadMemory { addr: ffd8, size: 01 }
Recv: [05, ff, d8, 01]
Recv: MemoryData(MemoryView { base: ffd8, data: [01] })
Sent: [05, ff, fe, 02]
Sent: ReadMemory { addr: fffe, size: 02 }
Recv: [05, ff, fe, 06, cb]
Recv: MemoryData(MemoryView { base: fffe, data: [06, cb] })
Sent: [40]
Sent: Unknown { cmd_id: 40, payload: [] }
Recv: [40, 15, 24]
Recv: Unknown { cmd_id: 40, payload: [40, 15, 24] }
Sent: [05, ff, e5, 01]
Sent: ReadMemory { addr: ffe5, size: 01 }
Recv: [05, ff, e5, 00]
Recv: MemoryData(MemoryView { base: ffe5, data: [00] })
Sent: [05, ff, e0, 01]
Sent: ReadMemory { addr: ffe0, size: 01 }
Recv: [05, ff, e0, 00]
Recv: MemoryData(MemoryView { base: ffe0, data: [00] })
Sent: [05, ff, da, 02]
Sent: ReadMemory { addr: ffda, size: 02 }
Recv: [05, ff, da, 5d, 01]
Recv: MemoryData(MemoryView { base: ffda, data: [5d, 01] })
Sent: [45, a1, 03, 00, 00, 02]
Sent: Unknown { cmd_id: 45, payload: [a1, 03, 00, 00, 02] }
Recv: [45, a1, 03, 00, 03, ff, b8, 00, 00, 00, 00]
Recv: Unknown { cmd_id: 45, payload: [45, a1, 03, 00, 03, ff, b8, 00, 00, 00, 00] }
Sent: [14, 00, 10, 02]
Sent: ReadFloats { addr: 10, len: 02 } (0x10 | 16) <> METER_D_IN_1
Recv: [14, 00, 10, 00, 00, f0, c2, 00, 00, f0, c2]
Recv: FloatData(FloatView { base: 10, data: [-120.0, -120.0] })
Sent: [14, 00, 10, 02]
Sent: ReadFloats { addr: 10, len: 02 } (0x10 | 16) <> METER_D_IN_1
Recv: [14, 00, 10, ff, 5d, 03, c3, ff, 5d, 03, c3]
Recv: FloatData(FloatView { base: 10, data: [-131.36717, -131.36717] })
Sent: [14, 00, 10, 02]
Sent: ReadFloats { addr: 10, len: 02 } (0x10 | 16) <> METER_D_IN_1
Recv: [14, 00, 10, 3e, 78, 03, c3, 3e, 78, 03, c3]
Recv: FloatData(FloatView { base: 10, data: [-131.4697, -131.4697] })
Sent: [14, 00, 10, 02]
Sent: ReadFloats { addr: 10, len: 02 } (0x10 | 16) <> METER_D_IN_1
Recv: [14, 00, 10, f3, 69, 03, c3, f3, 69, 03, c3]
Recv: FloatData(FloatView { base: 10, data: [-131.41386, -131.41386] })
Sent: [14, 00, 10, 02]
Sent: ReadFloats { addr: 10, len: 02 } (0x10 | 16) <> METER_D_IN_1
Recv: [14, 00, 10, 70, 8d, 03, c3, 70, 8d, 03, c3]
Recv: FloatData(FloatView { base: 10, data: [-131.55249, -131.55249] })
Sent: [14, 00, 10, 02]
Sent: ReadFloats { addr: 10, len: 02 } (0x10 | 16) <> METER_D_IN_1
Recv: [14, 00, 10, ef, 74, 03, c3, ef, 74, 03, c3]
Recv: FloatData(FloatView { base: 10, data: [-131.45677, -131.45677] })
Sent: [14, 00, 1c, 04]
Sent: ReadFloats { addr: 1c, len: 04 } (0x1c | 28) <> METER_OUT_1
Recv: [14, 00, 1c, 00, 00, f0, c2, 00, 00, f0, c2, 00, 00, f0, c2, 00, 00, f0, c2]
Recv: FloatData(FloatView { base: 1c, data: [-120.0, -120.0, -120.0, -120.0] })
Sent: [14, 00, 18, 04]
Sent: ReadFloats { addr: 18, len: 04 } (0x18 | 24) <> METER_COMP_1
Recv: [14, 00, 18, 00, 00, f0, c2, 00, 00, f0, c2, 00, 00, f0, c2, 00, 00, f0, c2]
Recv: FloatData(FloatView { base: 18, data: [-120.0, -120.0, -120.0, -120.0] })
Sent: [14, 00, 1c, 04]
Sent: ReadFloats { addr: 1c, len: 04 } (0x1c | 28) <> METER_OUT_1
Recv: [14, 00, 1c, 00, 00, f0, c2, 00, 00, f0, c2, 00, 00, f0, c2, 00, 00, f0, c2]
Recv: FloatData(FloatView { base: 1c, data: [-120.0, -120.0, -120.0, -120.0] })
Sent: [14, 00, 18, 04]
Sent: ReadFloats { addr: 18, len: 04 } (0x18 | 24) <> METER_COMP_1
Recv: [14, 00, 18, 00, 00, 00, 00, 00, 00, 00, 00, ff, ff, ff, ff, ff, ff, ff, ff]
Recv: FloatData(FloatView { base: 18, data: [0.0, 0.0, NaN, NaN] })
Sent: [14, 00, 1c, 04]
```

mrene, 2021-06-05:
These are consistent with the current implemention 🤔  the initial values are also -120dB - are you sure they aren't read properly?

mrene, 2021-06-05:
I added some [discord](https://discord.gg/XGHmrcDumf) links for 1:1 things like this, might be easier than github issues for debugging things

kukocz, 2021-06-06:
Good news are that after rebooting RPi and DDRC-24 it's working now.
API call:
```
{
  "master": {
    "preset": 1,
    "source": "Toslink",
    "volume": -22.0,
    "mute": false,
    "dirac": true
  },
  "input_levels": [
    -38.165657,
    -37.840828
  ],
  "output_levels": [
    -69.61046,
    -65.51071,
    -120.0
  ]
}
```
What is strange - I've got only 3 of 4 output readings.
The minidsp commands itself always returns -120db for input/output values:
```
$ minidsp
MasterStatus { preset: 1, source: Toslink, volume: Gain(-22.0), mute: false, dirac: true }
Input levels: -120.0, -120.0
Output levels: -120.0, -120.0, -120.0
```

mrene, 2021-06-06:
Interesting. The 3 channels is a bug in the way values are fetched, I'll fix that.

Are you saying that the input levels will suddenly read as -120dB if you wait long enough _between_ minidsp commands?

kukocz, 2021-06-07:
Hi,

Sometimes everything is working correctly, sometimes not. I couldn't determine conditions for which it behaves like I described.
Let's close this issue until I'm sure what's going on.

mrene, 2021-08-24:
Closing since this had no activity for a long time.

PR #143 (2021-06-05): Fix DDRC-24 support
Fixes #142
closed 2021-08-24 by mrene