mrene/minidsp-rs #155: Receiving frequent errors after upgrading to 0.1.1

mjlott, 2021-06-16:
Getting this error several times a night, which is only resolved with a hardware reset of the Mini DSP 2x4HD: error; EzBeq.getdevices failed, HTTP status 500. Log entry indicates the following: ProcessTimedOut: ('Process did not terminate within 10 seconds', ['/usr/bin/minidsp', '--tcp', '127.0.0.1']).

mrene, 2021-06-16:
Im assuming (because of the ezbeq reference) that you're running this on a raspberry pi and have installed the debian package, is that right?

When the problem occurs, are you able to run `minidsp` from the console?

If not, does `journalctl -u minidsp` show any errors?

mjlott, 2021-06-17:
Yes, I am running on a Raspberry Pi 4 and the Rasbian Pi (Debian) OS. When I got the issue, I lost connectivity and had to reboot the mDSP to get connectivity back. I reverted to version 0.0.8 earlier today so can’t run any commands to test. Sorry about that… Thanks for all your great work! I can’t tell you how much it means to be able to wirelessly access m mDSP from my laptop! It’s awesome! I never was able to get a wi-dg to work because its IP conflicted with my cable modem and I was stuck using usb until you came out with mini dsp-rs.

mrene, 2021-08-24:
Closing since this had no activity for a long time, and no other reports.

closed 2021-08-24 by mrene