mrene/minidsp-rs #191: [2021-08-16T18:50:05Z ERROR minidsp::transport::multiplexer] recv loop exit: ParseError(EmptyPacket)

sirjaymz, 2021-08-16:
C:\Calibration Automation\minidsp.x86_64-pc-windows-msvc>minidsp.exe probe
[2021-08-16T18:50:05Z ERROR minidsp::transport::multiplexer] recv loop exit: ParseError(EmptyPacket)
Found 2x4HD with serial 90#### at usb:%5C%5C%3F%5Chid%23vid_2752%26pid_0011%26rev_06f2%26mi_04%238%2622799767%260%260000%23%7B4d1e55b2-f16f-11cf-88cb-001111000030%7D?vid=2752&pid=0011 [hw_id: 10, dsp_version: 100]
Found 2x4HD with serial 90#### at usb:%5C%5C%3F%5Chid%23vid_2752%26pid_0011%26rev_06f2%26mi_04%238%26770d6d3%260%260000%23%7B4d1e55b2-f16f-11cf-88cb-001111000030%7D?vid=2752&pid=0011 [hw_id: 10, dsp_version: 100]

C:\Calibration Automation\minidsp.x86_64-pc-windows-msvc>

Not sure why I am getting the error.

version 0.1.2
I did have to write the actually serial numbers (redacted above) to the unit, as the standard 965535 serial was assigned to both of them. I assigned the actual serial number to both of them by disabling each unit with a USB tool in windows.

sirjaymz, 2021-08-17:
rebooted the machine. re-ran the same software, download again, thinking I might have a corrupt version, both with the same result. 
Tried moving the file to a non-space directory with same results.
I am connecting both of the units through an external Anker usb hub, and then to the laptop. 

I'll try connecting directly to machine, eliminating usb hub and see if that makes a difference.

sirjaymz, 2021-08-17:
Tried connecting minidsp 2x4hd directly to the laptop, eliminating the usb hub with no difference.. same error above still comes up..

So I tried going back to the previous version of minidsp-rs. 

This was the result of that...

C:\Calibration Automation\minidsp-rs\v0.1.1\minidsp.x86_64-pc-windows-msvc>minidsp.exe probe
thread 'tokio-runtime-worker' panicked at 'range start must not be greater than end: 1 <= 0', C:\Users\runneradmin\.cargo\registry\src\github.com-1ecc6299db9ec823\bytes-1.0.1\src\bytes.rs:245:9
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
Found 2x4HD with serial 90#### at usb:%5C%5C%3F%5Chid%23vid_2752%26pid_0011%26rev_06f2%26mi_04%238%26118c3fbf%260%260000%23%7B4d1e55b2-f16f-11cf-88cb-001111000030%7D?vid=2752&pid=0011 [hw_id: 10, dsp_version: 100]

C:\Calibration Automation\minidsp-rs\v0.1.1\minidsp.x86_64-pc-windows-msvc>

(serial number redacted)

mrene, 2021-08-17:
It's a bit weird you're getting this as part of a probe command, usually it happens when an unsupported device is used and and it doesn't support the command itself.

Can you run `minidsp -vvv probe` - that should show what's being sent on the wire.

mrene, 2021-08-24:
Are you still having that error? If so, could you run `minidsp -vvv probe` ?

mrene, 2021-09-23:
Closing due to inactivity.

closed 2021-09-23 by mrene