mrene/minidsp-rs #245: PWR-ICE125 / PWR-ICE250 device support?

SamHawkens, 2021-11-12:
Do you expect the PWR-ICE plate amplifiers to be supported?

https://www.minidsp.com/products/plate-amplifiers/pwr-ice125
https://www.minidsp.com/products/plate-amplifiers/pwr-ice250

They have a LAN control interface, no USB.
The datasheet states ADAU1445 SigmaDSP is used internally.

mrene, 2021-11-13:
At first glance the plugins look prety similar to some other product so it looks possible to add it. 

I don't see a field to connect to a specific IP directly, so I'd need a packet capture to see the kind of advertisements the device is sending out in order to be able to simulate it.

SamHawkens, 2021-11-13:
Sounds good. I have not bought the plate amps yet since I am also considering hypex fusion amps. But there does not seem to be any linux support at all. About the PWR-ICE I would guess they use zeroconf for detection. Or maybe they even use UDP for everything. At least they must have added some (most likely simple) uC to the ADAU1445. It comes a bit as a surprise that the plate amps do not have USB.

mrene, 2021-11-13:
Most of their devices use some rudimentary USB HID over TCP (on port 5333). The advertisement packets didn't look like anything to me (it's not mdns, and there's some kind of checksum I haven't figured out yet since I don't have a TCP only device)

m-diers, 2022-01-23:
I am currently also looking for a simple variant to volume control via lan for the PWR-ICE.My goal is to extend the alsa output in LibreELEC with it.
There is an android app for the minidsp.
https://m.apkpure.com/de/minidsp-controller/com.dsp4you.minidsp
What I saw at first sight is that written in js with angular and cordova. Just unzip the apk then there are the sources.
 I have already noticed the following: The pwr-ice also regularly gives a sign of life
```
Frame 2356: 116 bytes on wire (928 bits), 116 bytes captured (928 bits) on interface any, id 0
    Interface id: 0 (any)
        Interface name: any
    Encapsulation type: Linux cooked-mode capture v1 (25)
    Arrival Time: Jan 23, 2022 12:15:58.777236657 CET
    [Time shift for this packet: 0.000000000 seconds]
    Epoch Time: 1642936558.777236657 seconds
    [Time delta from previous captured frame: 0.280376643 seconds]
    [Time delta from previous displayed frame: 5.654376924 seconds]
    [Time since reference or first frame: 334.732787291 seconds]
    Frame Number: 2356
    Frame Length: 116 bytes (928 bits)
    Capture Length: 116 bytes (928 bits)
    [Frame is marked: False]
    [Frame is ignored: False]
    [Protocols in frame: sll:ethertype:ip:udp:data]
    [Coloring Rule Name: UDP]
    [Coloring Rule String: udp]
Linux cooked capture v1
    Packet type: Broadcast (1)
    Link-layer address type: Ethernet (1)
    Link-layer address length: 6
    Source: DSP4YOU_07:26 (00:50:c2:d4:37:26)
    Unused: 0000
    Protocol: IPv4 (0x0800)
Internet Protocol Version 4, Src: 192.168.178.120, Dst: 255.255.255.255
    0100 .... = Version: 4
    .... 0101 = Header Length: 20 bytes (5)
    Differentiated Services Field: 0x00 (DSCP: CS0, ECN: Not-ECT)
        0000 00.. = Differentiated Services Codepoint: Default (0)
        .... ..00 = Explicit Congestion Notification: Not ECN-Capable Transport (0)
    Total Length: 100
    Identification: 0x23d1 (9169)
    Flags: 0x00
        0... .... = Reserved bit: Not set
        .0.. .... = Don't fragment: Not set
        ..0. .... = More fragments: Not set
    ...0 0000 0000 0000 = Fragment Offset: 0
    Time to Live: 100
    Protocol: UDP (17)
    Header Checksum: 0xbf97 [validation disabled]
    [Header checksum status: Unverified]
    Source Address: 192.168.178.120
    Destination Address: 255.255.255.255
User Datagram Protocol, Src Port: 3999, Dst Port: 3999
    Source Port: 3999
    Destination Port: 3999
    Length: 80
    Checksum: 0xe529 [unverified]
    [Checksum Status: Unverified]
    [Stream index: 0]
    [Timestamps]
        [Time since first frame: 333.620134327 seconds]
        [Time since previous frame: 5.654376924 seconds]
    UDP payload (72 bytes)
Data (72 bytes)
    Data: 801205a000000050c2d437260000c0a8b2780003003300b60000000000005b0000000010…
    [Length: 72]
```
```
0000   00 01 00 01 00 06 00 50 c2 d4 37 26 00 00 08 00   .......P..7&....
0010   45 00 00 64 23 d1 00 00 64 11 bf 97 c0 a8 b2 78   E..d#...d......x
0020   ff ff ff ff 0f 9f 0f 9f 00 50 e5 29 80 12 05 a0   .........P.)....
0030   00 00 00 50 c2 d4 37 26 00 00 c0 a8 b2 78 00 03   ...P..7&.....x..
0040   00 33 00 b6 00 00 00 00 00 00 5b 00 00 00 00 10   .3........[.....
0050   50 57 52 2d 49 43 45 32 20 20 20 20 20 20 20 00   PWR-ICE2       .
0060   b8 05 d0 f4 4d c2 fa af 77 7c dc 0f 96 f4 b4 a5   ....M...w|......
0070   32 34 e4 87                                       24..
```
Until now I could always skip js ;-) Maybe we can work on it together.

SamHawkens, 2022-01-23:
Cool! I am surprised the sources are not obfuscated in the apk - nice :-)
I like the Idea of linking the pwr-ice up with alsa. I must admit I have never written anything in rust, but sooner or later I'll have to get started with it anyway. js is not my world either, but well - there are so many languages around you get in touch with all of them.

For the moment I have got the hypex fusion amp FA123 modules which seem much more popular and have better availablity at least around Europe. The only way to configure these from a linux machine is to use a windows vm with usb passthru in virtualbox. Not my preferred solution. So anything that natively works on linux would be a reason to get the pwr-ice instead. And the pwr-ice also provides much more dsp programming options.

mrene, 2022-01-23:
@m-diers these are broadcast packets sent by the devices so you can discover their ip. `minidsp probe --net` will listen and list them. The code to decode them are in [transport/net/discovery](https://github.com/mrene/minidsp-rs/blob/dev/minidsp/src/transport/net/discovery/mod.rs). This one expands to: `DiscoveryPacket { mac_address: [0, 80, 194, 212, 55, 38], ip_address: 192.168.178.120, hwid: 0, typ: 51, sn: 182, hostname: "PWR-ICE2       \u{0}" }`

The official mobile app only does a few basic commands, which are already supported by mindsp-rs - you should be able to use these same commands already, even if the PWR-ICE isn't directly supported.

ex: `minidsp --tcp 192.168.178.120` should give you the current master level, and `minidsp --tcp 192.168.178.120 probe` should tell you what device it's detecting (which would be a special `Generic` device for partially supported devices like this one).

[minidspd](https://minidsp-rs.pages.dev/daemon/http) will also automatically discover and expose devices that broadcast themselves on the network, this would give you an http api to control things (which will work for source selection, master volume, etc, but not for changing equalizers until the device support is added to the code)

m-diers, 2022-01-23:
@mrene This sounds great, I had tested it only once with status, and got no response.
```console
./minidsp --verbose probe --net
Probing for network devices...
Found: tcp://192.168.178.120:5333?name=PWR-ICE2%20%20%20%20%20%20%20%00
```
```console
./minidsp --verbose --tcp 192.168.178.120
Sent: [31]
Sent: ReadHardwareId
Recv: [31, 03, 00]
Recv: HardwareId { payload: [03, 00] }
Sent: [05, ff, a1, 01]
Sent: ReadMemory { addr: ffa1, size: 01 } 
Error: No devices found
```
```console
./minidsp --verbose --tcp 192.168.178.120 probe
Sent: [31]
Sent: ReadHardwareId 
Recv: [31, 03, 00]
Recv: HardwareId { payload: [03, 00] }
Sent: [05, ff, a1, 01]
Sent: ReadMemory { addr: ffa1, size: 01 }
```
Unfortunately it doesn't work yet. I'll check it out now and compile it myself.

mrene, 2022-01-23:
I'm guessing it's timing out after that last `ReadMemory` ?

Any chance you could packet capture the plugin connecting to the device? (it'll be on tcp port 5333)

m-diers, 2022-01-23:
Yes it's a timeout.
I have wireshark running here. Which format is desired?
https://gist.github.com/m-diers/7aaee0444c1baee4f3423887704e951a
https://gist.github.com/m-diers/42415734297a506b58da06b07b4cf673
FYI: The special executable to configure the PWR-ICE has the name PWR_ICE2_2x2.exe

mrene, 2022-01-23:
pcap (or pcapng) would be ideal

mrene, 2022-01-23:
Just to clarify, a pcap of the plug-in is what I'm looking for (since minidsp-rs apparently doesn't do the right thing)

m-diers, 2022-01-23:
Here some captures:
`./target/release/minidsp --verbose --tcp 192.168.178.120`
https://gist.github.com/m-diers/0b06f05ab6d71a3706be2158c39d63c6#file-capture-001-pcapng
`./target/release/minidsp --verbose --tcp 192.168.178.120 probe`
https://gist.github.com/m-diers/0b06f05ab6d71a3706be2158c39d63c6#file-capture-002-pcapng
`Connect of PWR_ICE2_2x2`
https://gist.github.com/m-diers/0b06f05ab6d71a3706be2158c39d63c6#file-capture-003-pcapng

mrene, 2022-01-23:
Thanks!

It appears that the framing is a bit different as the device sends fixed-sized 70 bytes frames instead of fixed-size 64 bytes frames, which confuses the code handling the transport - there's probably a quick fix for this I'll take a look at it.

mrene, 2022-01-23:
@m-diers are you able to try #313 and let me know if it can connect to your device?

m-diers, 2022-01-24:
Thanks!
the connect works.
Now there are probably still problems when reading the appropriate register for the I/O levels.
Master Mute is ok:
```console
MasterStatus { preset: 0, source: NotInstalled, volume: Gain(-0.0), mute: false, dirac: false }
#toggled with app
MasterStatus { preset: 0, source: NotInstalled, volume: Gain(-0.0), mute: true, dirac: false }
```
Master Volume I can not test because it does not work via the app. Maybe it is not supported at all.
I have found more information in the logs of the app, I will summarize them.