mrene/minidsp-rs #282: Connection to mnidsp 2x4HD breaks after using IR remote

ismal, 2021-12-20:
Hi!

I installed minidsp-rs on my old C.H.I.P. computer which which is a kind of raspberry pi that runs debian buster now and everything works good, I can connect to my 2x4HD using the original GUI and do all the changes, for example update and change profiles. BUT, if I start changing profiles with the IR Remote control while connected to the GUI using minidsp-rs, I see the change of a Config profile but then I get a dialog that the connection was broken. In the logs I see this: 

Dec 20 18:28:13 chip systemd[1]: Started MiniDSP Server.
Dec 20 18:28:13 chip minidspd[1285]: App is listening on: /tmp/minidsp.sock
Dec 20 18:28:14 chip minidspd[1285]: App is running on: 0.0.0.0:5380
Dec 20 18:28:44 chip minidspd[1285]: [2021-12-20T17:28:44Z ERROR minidsp::transport::hub] device_tx: An HID error has occurred: hidapi error: hid_error

ismal, 2021-12-20:
It only happens after switching Configs, switching volume level or source does not break the server. Here are some output with -v option 

Recv: MemoryData(MemoryView { base: ffe0, data: [00] })
Sent: [05, ff, da, 02]
Sent: ReadMemory { addr: ffda, size: 02 }
Recv: [05, ff, da, 11, 00]
Recv: MemoryData(MemoryView { base: ffda, data: [11, 00] })
Recv: [05, ff, d8, 00]
Recv: MemoryData(MemoryView { base: ffd8, data: [00] })
[2021-12-20T17:47:08Z ERROR minidsp::transport::hub] device_tx: An HID error has occurred: hidapi error: hid_error is not implemented yet
Recv: [05, ff, d8, 00]
Recv: MemoryData(MemoryView { base: ffd8, data: [00] })

mrene, 2022-01-17:
It seems that the device stops accepting usb commands while the configuration is being switched. I've upped the number of retries on usb failures to 10 which gives it sufficient time to start accepting commands again (it retries twice here).

PR #305 (2022-01-17): fix(hid): retry failed writes more often
fixes #282
commit f6f945f5 (2022-01-17): fix(hid): retry failed writes more often. fixes #282
closed 2022-01-17 by mrene