mrene/minidsp-rs #370: miniDSP Flex Support

Mobe1969, 2022-05-09:
Is it possible to get miniDSP Flex support working? 

I'm moberhardt on AVSForum and runs one of the filter repositories.

The flex   uses the same plugin as 2x4HD, same specs channel and filter wise, and loads the XML files fine

Running basic command line sort of works, but doesn't show source/input:
MasterStatus { preset: 1, source: NotInstalled, volume: Gain(-0.0), mute: false, dirac: false }
Input levels:
Output levels:

Loading a filter via ezbeq doesn't work though

`During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "ezbeq\apis\ws.py", line 111, in __send_to_all
  File "autobahn\websocket\protocol.py", line 2254, in sendMessage
autobahn.exception.Disconnected: Attempt to send on a closed protocol
2022-05-09 16:39:04,984 - PoolThread-twisted.internet.reactor-3 - ezbeq.ws - INFO - unregister - Unregistering client tcp4:192.168.0.16:50614
2022-05-09 16:39:04,984 - PoolThread-twisted.internet.reactor-3 - ezbeq - ERROR - log_exception - Exception on /api/2/devices/master [PATCH]
Traceback (most recent call last):
  File "flask\app.py", line 1516, in full_dispatch_request
  File "flask\app.py", line 1502, in dispatch_request
  File "flask_restx\api.py", line 403, in wrapper
  File "flask\views.py", line 84, in view
  File "flask_restx\resource.py", line 49, in dispatch_request
  File "ezbeq\apis\devices.py", line 303, in patch
  File "ezbeq\device.py", line 161, in update
  File "ezbeq\minidsp.py", line 597, in update
  File "ezbeq\device.py", line 254, in _hydrate_cache_broadcast
  File "ezbeq\minidsp.py", line 585, in __do_it
  File "ezbeq\minidsp.py", line 624, in __update_slot
  File "ezbeq\minidsp.py", line 538, in set_gain
  File "ezbeq\device.py", line 254, in _hydrate_cache_broadcast
  File "ezbeq\minidsp.py", line 535, in __do_it
  File "ezbeq\minidsp.py", line 439, in __send_cmds
  File "concurrent\futures\_base.py", line 444, in result
  File "concurrent\futures\_base.py", line 389, in __get_result
  File "concurrent\futures\thread.py", line 57, in run
  File "ezbeq\minidsp.py", line 562, in __do_run
  File "plumbum\commands\base.py", line 240, in run
  File "plumbum\commands\base.py", line 201, in runner
  File "plumbum\commands\processes.py", line 322, in run_proc
  File "plumbum\commands\processes.py", line 24, in _check_process
  File "plumbum\machines\base.py", line 28, in verify
plumbum.commands.processes.ProcessExecutionError: Unexpected exit code: 1
Command line: | minidsp.exe -f 'C:\Users\oberh\AppData\Local\Temp\tmpsapvg49q'
Stderr:       | Error: Specified channel or peq is out of range`

If there is anything I can do to help diagnose it or assist let me know.

Mobe1969, 2022-05-09:
Note I also do have a wi-dg (not in use) that I could possibly use if there may be some way of capturing at the TCP level from the plugin?

Mobe1969, 2022-05-09:
[minidsp flex.zip](https://github.com/mrene/minidsp-rs/files/8650606/minidsp.flex.zip)
I've attached a capture file with Microsoft Network Monitor showing the data when going to the flex via a WIDG.

adamcstephens, 2022-05-20:
I just received a Flex and would be happy to help. Here's the probe output

```
minidsp probe
Found Generic with serial 901426 at usb:5-3.1.2%3A1.4?vid=2752&pid=0011 [hw_id: 27, dsp_version: 100]
```

adamcstephens, 2022-05-20:
I have a Flex balanced without Dirac. Here's the config

[config.xml.txt](https://github.com/mrene/minidsp-rs/files/8735937/config.xml.txt)

closed 2022-05-21 by mrene

mrene, 2022-05-21:
[v0.1.5 ](https://github.com/mrene/minidsp-rs/releases/tag/v0.1.5) has been released, including Flex support added by @adamcstephens

PR #380 (2022-05-21): initial flex support
I'm mostly just following the existing pattern. The codegen mod.rs file is a copy of 2x4hd, which may or may not match all the options of the flex. If you have any pointers on how to verify the input/output options, please let me know.

If I missed anything or did anything wrong, please point it out.

resolves: #370