mrene/minidsp-rs #379: illegal hardware instruction error with

alex-vyverman, 2022-05-20:
Hi!
Very grateful for your application.
I'm currently trying to create a Stream Deck plugin that talks to the HTTP API on minidspd, but I'm often faced with the daemon crashing with the error `zsh: illegal hardware instruction `.
Is this a known issue?
I'm running minidsp-daemon 0.1.4

Many thanks!
Alex

alex-vyverman, 2022-05-20:
I'm running on MacOS Big Sur 11.6.5

mrene, 2022-05-21:
Hi!

Not a known issue at all.

Can you tell me what kind of mac are you running this on? Is it intel or apple silicon? Are you buildling yourself or using one of the pre-built binaries?

Any steps to reproduce the crash, or does it happen randomly? 

If you can run this with a debugger, the stack trace will be essential to point out where this happens.

closed 2022-10-09 by mrene