mrene/minidsp-rs #383: MiniDSP Flex Input Switching Delay

Mobe1969, 2022-05-31:
Just a minor issue with ending command sequences to the flex that start with an input switch, the Flex is somewhat slower than the 2x2HD and takes approximately 4-5 seconds. So if you are say using ezBEQ and selecting a slot to load a filter into, it doesn't load. @3ll3d00d at avsforum who does ezbeq (https://github.com/3ll3d00d/ezbeq) said it would!d need to be changed on the minidsp rs side.

Thanks

mrene, 2022-06-04:
That's odd because the device actually sends a confirmation that the active configuration has switched, and minidsp-rs waits for it before issuing further commands.

It's entirely possible that the device reports the config change too soon, and that this would be patched by adding some artificial delay 

Can you try the following:
- Play some audio through the device
- Run the command below (it'll move to preset 1, and mute both input channels)
- Tell me if the audio stops

```
echo -e "config 1\ninput 0 mute on\ninput 1 mute on\n" | minidsp -vvv -f -
```

To unmute the input channels again:

```
minidsp input 0 mute off
minidsp input 1 mute off
```

3ll3d00d, 2022-06-04:
I added a configurable delay of this nature (i.e. between config and subsequent commands) to ezbeq so hopefully someone will test the above or using that feature and report back (https://www.avsforum.com/threads/ezbeq-use-and-development-discussion.3181732/page-139#post-61725110)

mrene, 2022-06-05:
There is a test build [here](https://github.com/mrene/minidsp-rs/actions/runs/2443465884) which waits for the device to report the new preset before completing the config switch. 

I'm not sure whether the Flex reports the right configuration at first, but it would be a better solution than hardcoded delays, if it works.

Feel free to report the device's behaviour based on my first comment if you own a Flex device.

mrene, 2022-06-07:
The tentative fix worked, I'll get a release going soon.

Relaying some info from Discord in case it's useful to someone - the device immediately acknowledges the `SetConfig` command but still reports the previous preset as active - the next query (500ms later) is responded once the preset switch has been  fully executed.

```
Preset change sent
[0.016s] Sent: [25, 01, 01]
[0.016s] Sent: SetConfig { config: 01, reset: true }

Preset change acknowledged (too soon)
[0.019s] Recv: [ab]
[0.019s] Recv: ConfigChanged

Ask the device for its current preset
[0.019s] Sent: [05, ff, d8, 09]
[0.020s] Sent: ReadMemory { addr: ffd8, size: 09 }

Device responds with the previous preset
[0.022s] Recv: [05, ff, d8, **00**, 00, 00, 00, ff, ff, ff, ff, 00]
[0.022s] Recv: MemoryData(MemoryView { base: ffd8, data: [**00**, 00, 00, 00, ff, ff, ff, ff, 00] })

Ask again
[0.523s] Sent: [05, ff, d8, 09]
[0.523s] Sent: ReadMemory { addr: ffd8, size: 09 }

Device responds with the new preset, ~4 seconds later
[4.688s] Recv: [05, ff, d8, **01**, 00, 00, 00, ff, ff, ff, ff, 00]
[4.689s] Recv: MemoryData(MemoryView { base: ffd8, data: [**01**, 00, 00, 00, ff, ff, ff, ff, 00] })
...
```

mrene, 2022-06-07:
Fixed in [v0.1.7](https://github.com/mrene/minidsp-rs/releases/tag/v0.1.7)

closed 2022-06-07 by mrene