mrene/minidsp-rs #400: Config Change Detection

giomania, 2022-06-24:
I am using EZBEQ and when I change configurations using the remote on the mini DSP device this change is not detected. When I change volume using the mini DSP remote this change is detected. I am using a DDRC-88A, if that is a factor.

Thank you.

mrene, 2022-07-31:
I seem to have missed this. 

Could you run `minidsp -vvv server` and change the configuration via the remote? It'll output the commands that are sent by your device.

giomania, 2022-07-31:
Hi Mathieu, 

 

Just yesterday, a friend of mine that is more tech-savvy than I set up a new miniDSP 2x4 HD. He confirmed that miniDSP-RS was reading the configuration preset changes in his setup with the miniDSP 2x4 HD, and helped me confirm that it was also working in my setup with the miniDSP 88A. Subsequently, he also set up EZBEQ and confirmed that it is not refreshing the configuration preset changes.

 

So it would appear this is an issue with EZBEQ, vice miniDSP-RS.

 

Thank you.

 

Mark

mrene, 2022-07-31:
Thanks, closing as such!

closed 2022-07-31 by mrene