mrene/minidsp-rs #511: Not showing in the app after bootup until service is restarted

gwurb, 2022-12-12:
MiniDSP device not showing in the app after bootup. Have to restart the service or add a sleep timer.

Setup:
minidsp-rs installed on volumio 3. Volumio 3 is running on raspberry pi zero 2 w using a usb dongle. 
/etc/minidsp/config.toml has
bind_address = "0.0.0.0:5380"
bind_address = "0.0.0.0:5333"
advertise = { ip = "192.168.1.20", name="Volumio" } 

where 192.168.1.20 is the DHCP reservation for the raspberry pi 2 w.

After bootup:
-bash-5.0# systemctl status minidsp
minidsp.service - MiniDSP Server
   Loaded: loaded (/lib/systemd/system/minidsp.service; enabled; vendor preset: enabled)
   Active: active (running) since Mon 2022-12-12 08:49:49 UTC; 1min 43s ago

But, minidsp device does not show up in the app.

If the service is restarted with:
systemctl restart minidsp

minidsp device does show up in the app.

Changing /lib/systemd/system/minidsp.service to the following does not help
[Unit]
After=network-online.target
Wants=network-online.target

Had to add 
[Service]
ExecStartPre=/bin/sleep 30

Is there any other way to solve the device not showing in the app without a sleep pre service start?

mrene, 2022-12-17:
I wonder if it wouldn't be binding to the wrong address when sending out broadcast packets, can you try setting the config to this?

```
advertise = { ip = "192.168.1.20", name="Volumio", bind_address="192.168.1.20" }
```

You can also set the `RUST_LOG` environment var to `trace` to get (a ton more) debug information.

gwurb, 2022-12-18:
I tried:
> advertise = { ip = "192.168.1.20", name="Volumio", bind_address="192.168.1.20" }
but that made things worse. No device in the app, even after service restart. Had to delete the bind_address potion of that line (roll back) to get the service to work.

I now changed to a USB Eth adapter so the network interface should be coming up faster than WiFi.

Still, issue persists. 

I don't know RUST but am willing to try the RUST_LOG suggestion. Where do I set it? In a file or command line? And, where will it log to?

mrene, 2022-12-20:
What most likely is happening is that minidspd is trying to bind on startup, and not retrying it later if it fails (although I'm confused as to why this happens even if you have `Wants=network-online.target` in your systemd unit file). The bind address I told you to set probably need a port number as well, or at least `:0` to indicate it should use any available port (`bind_address="192.168.1.20:0"`).

`minidspd` could retry to setup its advertisement broadcast if it detects a failure, that seem like it would solve the problem

kettenbach-it, 2023-03-12:
Same here. Adding a port ":0" doesn't help. minidsp always needs a scheduled restart, to be announced.

kettenbach-it, 2023-03-12:
Fixed it. You need to set:

```
[Unit]
After=default.target
```

instead of `network-target`

scottshanafelt, 2023-03-25:
Also experiencing this issue with default service config

SlaveUnit22, 2023-04-19:
I'm also having the same issue as the OP. This change did not change anything for me unfortunately:
[Unit]
After=default.target

MervisDiamonds, 2023-05-13:
I have the same problem too.