mrene/minidsp-rs #516: ERROR minidsp::transport::multiplexer] recv loop exit: ParseError(EmptyPacket)

conrad-nash, 2022-12-20:
I have a pi3 running minidsp-rs connected to a minidsp 2x4HD. It runs for days/weeks without issue and then I start getting the following error:

`pi@singlesubs:~ $ minidsp
[2022-12-20T19:55:24.732Z ERROR minidsp::transport::multiplexer] recv loop exit: ParseError(EmptyPacket)
[2022-12-20T19:55:24.732Z ERROR minidsp::transport::multiplexer] recv loop exit: ParseError(EmptyPacket)
Error: A device request timed out
`

I have a second pi connected to a different minidsp 2x4HD and that's working fine. I suspect it's a minidsp issue but I wanted to flag it here just in case it's useful.

I saw another post where the user had the same issue and you asked him to run a specific command. I've followed that and it looks like minidsp-rs can see the minidsp.

`pi@singlesubs:~ $ minidsp -vvv probe
[0.000s] Sent: [31]
[0.000s] Sent: ReadHardwareId 
[0.005s] Recv: [31, 01, 11, 0a]
[0.005s] Recv: HardwareId(HardwareId { fw_major: 01, fw_minor: 11, hw_id: 0a })
[0.006s] Sent: [05, ff, a1, 01]
[0.006s] Sent: ReadMemory { addr: ffa1, size: 01 } 
[0.008s] Recv: [05, ff, a1, 64]
[0.009s] Recv: MemoryData(MemoryView { base: ffa1, data: [64] })
[0.010s] Sent: [05, ff, fe, 02]
[0.010s] Sent: ReadMemory { addr: fffe, size: 02 } 
[0.012s] Recv: [05, ff, fe, 4f, c7]
[0.013s] Recv: MemoryData(MemoryView { base: fffe, data: [4f, c7] })
0: Found 2x4HD with serial 920423 at ws://localhost/devices/0/ws [hw_id: 10, dsp_version: 100]
`

Is there anything further I can do to confirm where the issue lies? I've rebooted the pi and the issue persists. I can reboot the minidsp as well but I thought it might be useful to investigate while the error is happening.

conrad-nash, 2022-12-20:
An update to this. I've been able to resolve the error. If I connect to the minidsp using the official plugin (but using the pi IP) I can connect but I see the VU meters maxxed out and static. I assume this is because the plugin uses the same api that minidsp-rs uses and that's not responding with an expected response.

If I change configs then everything starts working again, including the config that was previously not working. Minidsp-rs now sees the expected response, the VUs in the plug in work properly.

So I'm pretty certain that it's a minidsp issue and not a minidsp-rs issue. Feel free to close this unless there's anything useful for you.

mrene, 2022-12-20:
>  I see the VU meters maxxed out and static
I've reached this state before when developing things and got the minidsp into a werid state, it always came back to normal after a power cycle, and it's been very random. If you have way of reproducing this state I'd be curious.

The `EmptyPacket` error is typically indicative of an unsupported command, which sounds weird in this case, but if the device isn't responding to the official app properly, I would assume that it's mostly because the device is in a broken state.

conrad-nash, 2022-12-21:
I'll see if I can recreate the issue. I have a hunch that it's something to do with using ezBEQ and the plugin in quick succession but I'll need to test that.

I'll close this now but I'll reference it if I find anything out.

If it happens again I'll try using minidsp-rs to change the config and see if that helps or resets things.

closed 2022-12-21 by conrad-nash