mrene/minidsp-rs #549: Is it possible to automatically overflow crossover PEQ?

chadrockey, 2023-02-10:
I'm looking to use the MSO strategy of loading input PEQs into the output channel PEQ and output channel specific PEQ into the crossover slots.

The files exported don't account for the two groups of four crossovers.  Is there a way to enable automatic overflow of the first crossover group into the second?  The error printed is:

` Warning: Some filters were not imported because they didn't fit (try using `all`) `
 
 so some user convenience and potentially dropping the need to specify groups would be great.

I'm really hoping to load a set of 8 PEQ into the crossover slots all at once.  This is possible with copy paste through the GUI, so I'm hoping we can make it work here.

mrene, 2023-02-12:
To stay backwards compatible, I added an `all` value for the group parameter which allows you to use import across both groups simultaneously.

Implemented in def74a7

```
$ minidsp output 0 crossover all all import /home/mrene/dev/minidsp-rs/minidsp/test_fixtures/rew-filters.txt
Xover 0.0: Applied imported filter: biquad1
Xover 0.1: Applied imported filter: biquad2
Xover 0.2: Applied imported filter: biquad3
Xover 0.3: Applied imported filter: biquad4
Xover 1.0: Applied imported filter: biquad5
Xover 1.1: Applied imported filter: biquad6
Xover 1.2: Applied imported filter: biquad7
Xover 1.3: Applied imported filter: biquad8
```

closed 2023-02-12 by mrene

chadrockey, 2023-02-13:
Amazing!  This should really simplify how to use these filters since they are so critical with BEQ.