mrene/minidsp-rs #565: input and output commands don't work

dgrzeszczak, 2023-03-11:
I've tried to use input and output command to change routing or mute outputs from command line on DDRC-24 with firmware version 1.17. Commands do not make effects even I restart and reconnect to minds using their app on MAC. Other commands like source, config work like a charm.

dgrzeszczak, 2023-03-11:
small update, it look like it works but the effect is not visible for the app (I'm using it on Mac) even after force closing the app

mrene, 2023-03-11:
The official applications do not read any state from the device (most devices don't support reading back their configuration), they read it from local files where the application is installed. The commands sent are the same as the application and affect the runtime state of the devices directly. 

Only some global state (master gain, mute, source, preset) will be reflected in the app.

closed 2023-03-11 by mrene

dgrzeszczak, 2023-03-12:
ok, thank you for clarification. Are there any other (unofficial) apps that handle that ?

mrene, 2023-03-12:
It's a device limitation really, no apps can read the full current config from a device. I don't know of any apps that maintain the same xml configuration as the official apps in order to interchange state.