mrene/minidsp-rs #566: miniDSP Flex Eight support

florentin, 2023-03-12:
Is it possible to add full support for Minidsp Flex Eight?
https://www.minidsp.com/products/minidsp-in-a-box/minidsp-flex-eight

These commands don't work as expected:
1. `minidsp source toslink`
Whatever value i use for the source (`  <VALUE>  The source to use: analog, toslink, spdif, usb, aesebu, rca, xlr, lan, i2s, bluetooth`) the device switches to `Analog`. Flex Eight doesn't have an analog input.

2. `minidsp status`
MasterStatus { preset: 1, source: NotInstalled, volume: Gain(-3.0), mute: false, dirac: false }
Input levels: 
Output levels: 

In this case, Dirac should have been true (it was enabled)
Input and Output levels were not detected.

This is the probe output:
`minidsp probe`
0: Found Generic with serial 901224 at ws://localhost/devices/0/ws [hw_id: 30, dsp_version: 111]

Other commands worked as expected: gain, mute, config, dirac

Personally, I need the `minidsp source` command to work. I own the device and can help with running anything you need.

mrene, 2023-03-12:
Does the Flex eight have any other sources than the flex? If it's the same I would assume that they would use similar identifiers, and you would be able to pretend that it's a 4 output Flex.

You could try `minidsp --force-kind flex source usb`. I wouldn't use this with inputs/outputs but it should be relatively safe for global settings like source, gain, mute, config, dirac.

Could you post the xml configuration file generated by the app? (the legacy one, not the device console, as its changed format and the scripts in the repository will need more work to be adapted)

mrene, 2023-03-19:
I only have the device console on my side, usually there is another app but maybe they didn't build one for the Flex Eight since it's new.

I tried simulating the device but the console is sending an unknown command I don't know, would you run this and let me know what it outputs?

```
minidsp -vvv debug send 40
```

florentin, 2023-03-19:
```bash
pi@raspberrypi:~ $ minidsp -vvv debug send 40
[0.000s] Sent: [31]
[0.001s] Sent: ReadHardwareId 
[0.005s] Recv: [31, 01, 37, 1e]
[0.007s] Recv: HardwareId(HardwareId { fw_major: 01, fw_minor: 37, hw_id: 1e })
[0.010s] Sent: [05, ff, a1, 01]
[0.012s] Sent: ReadMemory { addr: ffa1, size: 01 } 
[0.014s] Recv: [05, ff, a1, 6f]
[0.015s] Recv: MemoryData(MemoryView { base: ffa1, data: [6f] })
[0.017s] Sent: [05, ff, fe, 02]
[0.019s] Sent: ReadMemory { addr: fffe, size: 02 } 
[0.021s] Recv: [05, ff, fe, 04, c8]
[0.023s] Recv: response: Unknown { cmd_id: 40, payload: [40, 17, 24] }
```

I have managed to change the source on Flex Eight by using these commands:
```bash
minidsp debug send 3401 # - toslink
minidsp debug send 3402 # - spdif
minidsp debug send 3403 # - usb
```

I've discovered 0x34 by checking out the node-minidsp project: https://github.com/mrene/node-minidsp/blob/master/src/device.js#L144

Here's the debug for 0x34
```bash
pi@raspberrypi:~ $ minidsp -vvv debug send 34
[0.000s] Sent: [31]
[0.001s] Sent: ReadHardwareId 
[0.004s] Recv: [31, 01, 37, 1e]
[0.006s] Recv: HardwareId(HardwareId { fw_major: 01, fw_minor: 37, hw_id: 1e })
[0.007s] Sent: [05, ff, a1, 01]
[0.008s] Sent: ReadMemory { addr: ffa1, size: 01 } 
[0.009s] Recv: [05, ff, a1, 6f]
[0.010s] Recv: MemoryData(MemoryView { base: ffa1, data: [6f] })
[0.012s] Sent: [05, ff, fe, 02]
[0.013s] Sent: ReadMemory { addr: fffe, size: 02 } 
[0.014s] Recv: [05, ff, fe, 04, c8]
[0.016s] Recv: MemoryData(MemoryView { base: fffe, data: [04, c8] })
[0.017s] Sent: [34]
[0.018s] Decode error: DecodeError(InvalidLength { remaining: 0, required: 1 })
response: Ack
[0.066s] Recv: []
```

In case it makes any difference, i'm running those commands from a Raspberry Pi 3.

mrene, 2023-03-19:
I got the same results on the device console, it seems like the Flex source mappings are working for the Flex Eight, but the `--force-kind` option doesn't work for sources because of some other bug (bluetooth is 04).

In the new codebase, that command is defined here:
https://github.com/mrene/minidsp-rs/blob/094e8717b306c7e2e239f26059b4bb0495d557c4/protocol/src/commands.rs#L299-L302

https://github.com/mrene/minidsp-rs/blob/094e8717b306c7e2e239f26059b4bb0495d557c4/protocol/src/commands.rs#L537-L540

conrad-nash, 2023-11-26:
Hi, Is there any update on this? I have just picked up a flex 8 and I'm getting an error of "Error: Specified channel or peq is out of range" when I try and change the PEQ through minidsp-rs.

Kriegbaum, 2023-12-02:
I have also just picked up a flex eight, is there anything I can do to help bring it up to a fully supported device?

sameert123, 2023-12-05:
Also adding onto this - would love to help out in any way possible

sameert123, 2024-02-15:
Howdy - just wanted to bump this @mrene :) ty for this wonderful resource btw

PalmDK, 2024-05-30:
I made a bit of a dirty solution for this. I only needed source, volumen and preset so thats what i have been testing.
I have the FlexHTx, so it has a bit different inputs.

But i changed:
/protocol/src/device/probe.rs
Under line 76 i added my hw_id and dsp_version and made it show as a Flex (Running probe will tell you this info). I dont really think this is needed, but didnt test without it.

/protocol/src/source.rc
I added my hw_id (32) and the inputs of the FlexHTx below line 60. I needed to add Hdmi to the enum also as none of the others has Hdmi.
32 => &[
                (Analog, 0),
                (Toslink, 1),
                (Spdif, 2),
                (Usb, 3),
                (Hdmi, 4),
            ],

This works with the FlexHTx and should also with the Flex Eigth if you change to the correct inputs and hw / dsp id.