mrene/minidsp-rs #568: Check with DeviceControl Software Update

holisticagile, 2023-04-15:
Hey,

did anybody check compatibility with new Devices Control Software from minidsp?
Would love to clarify before updating and breaking the connection for minidsp-rs
thy

scottshanafelt, 2023-04-21:
I think you are refering to device console?  I did upgrade my SHD to that latest firmware.  While minidsp-rs works in every way I've tried with it, the ability for device console to "tunnel" through minidsp-rs directly to the SHD does not work.  Device Console throws some error when I attempt to connect.  The MiniDSP android app does continue to work.

holisticagile, 2023-04-24:
ok, cool. Thank you for your reply.
I would not like to loose the ability to control via. api my minidsp. :)

closed 2023-04-24 by holisticagile