mrene/minidsp-rs #621: Floating Point Error on new rpiOS Bookworm

kaelaria, 2023-10-20:
On the new Raspberry PI OS 'Bookworm', on a fresh install, minidsp gives a notice/warning during install and a floating point error when trying to run. Flashing back to 'Bullseye" still works fine.

3ll3d00d, 2024-03-28:
appears to be an issue with the armhf build installed on either a 32bit OS or a 64bit OS with arm32 added via multiarch, the 64bit build installed on bookwork 64bit is fine

3ll3d00d, 2024-03-29:
same applies when run on a 32bit OS with a 32 bit cpu (e.g. pi zero), looks like that package is just broken now

mrene, 2024-03-29:
> same applies when run on a 32bit OS with a 32 bit cpu (e.g. pi zero), looks like that package is just broken now

@3ll3d00d are you able to dump a stack trace and disassembly of where the invalid FP operation happens?

3ll3d00d, 2024-03-30:
I'll see what I can do

3ll3d00d, 2024-03-31:
not sure it helps as the trace seems corrupt and I'm not sure how to dig into that further

```
#0  0xf7be9a24 in ?? () from /lib/arm-linux-gnueabihf/libc.so.6
#1  0xf7ba1cfc in raise () from /lib/arm-linux-gnueabihf/libc.so.6
#2  0xf7cb9214 in ?? () from /lib/arm-linux-gnueabihf/libc.so.6
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
```

I decided to try bumping all the versions to current and seeing if I could get it to build given the mention of libc6 and looking at the current rpi build script/docker (which uses a variety of things that seem retired/legacy). This does produce a functioning binary on bookworm, by functioning I mean "doesn't crash immediately" i.e. I haven't tested it beyond that. You can find the changes in this diff https://github.com/mrene/minidsp-rs/compare/dev...3ll3d00d:minidsp-rs:dev

closed 2025-06-27 by mrene