mrene/minidsp-rs #625: The subcommand "peq x set x x x x x" does not accept negative values

hhartmut, 2023-11-26:
### Problem
The subcommand  input x peq x set x does not accept negative values. I have tried various forms, but there is always a different error message:

```
C:\workspace\minidsp-rs>minidsp -v --tcp 127.0.0.1 input 0 peq 0 set 1.0 -2.0 3.0 4.0 -5.0
error: Found argument '-2' which wasn't expected, or isn't valid in this context

  If you tried to supply '-2' as a value rather than a flag, use '-- -2'

Usage: minidsp input peq <INDEX> set [COEFF]...

For more information try '--help'

C:\workspace\minidsp-rs>minidsp -v --tcp 127.0.0.1 input 0 peq 0 set 1.0 -- -2.0 3.0 4.0 -- -5.0
error: Invalid value '--' for '[COEFF]...': invalid float literal

For more information try '--help'

C:\workspace\minidsp-rs>minidsp -v --tcp 127.0.0.1 input 0 peq 0 set 1.0 --2.0 3.0 4.0 --5.0
error: Found argument '--2.0' which wasn't expected, or isn't valid in this context

  If you tried to supply '--2.0' as a value rather than a flag, use '-- --2.0'

Usage: minidsp input peq <INDEX> set <COEFF>

For more information try '--help'
```

It only is successful with positive values:

```
C:\workspace\minidsp-rs>minidsp -v --tcp 127.0.0.1 input 0 peq 0 set 1.0 2.0 3.0 4.0 5.0
[0.000s] Sent: [31]
[0.001s] Sent: ReadHardwareId
[0.059s] Recv: [31, 01, 0d, 0a]
[0.062s] Recv: HardwareId(HardwareId { fw_major: 01, fw_minor: 0d, hw_id: 0a })
[0.063s] Sent: [05, ff, a1, 01]
[0.064s] Sent: ReadMemory { addr: ffa1, size: 01 }
[0.121s] Recv: [05, ff, a1, 64]
[0.123s] Recv: MemoryData(MemoryView { base: ffa1, data: [64] })
[0.124s] Sent: [05, ff, fe, 02]
[0.125s] Sent: ReadMemory { addr: fffe, size: 02 }
[0.184s] Recv: [05, ff, fe, 2b, 67]
[0.186s] Recv: MemoryData(MemoryView { base: fffe, data: [2b, 67] })
[0.187s] Sent: [30, 80, 20, 85, 00, 00, 00, 00, 80, 3f, 00, 00, 00, 40, 00, 00, 40, 40, 00, 00, 80, 40, 00, 00, a0, 40]
[0.190s] Sent: WriteBiquad { addr: Addr { val: 2085, len: 03, extra_bit: true }, data: [Value { Bytes: [0, 0, 80, 3f] (i: 0 | i32: 32831 | f32: 1.0 | fp: FixedPoint(0.00391376)) }, Value { Bytes: [0, 0, 0, 40] (i: 0 | i32: 64 | f32: 2.0 | fp: FixedPoint(7.6293945e-6)) }, Value { Bytes: [0, 0, 40, 40] (i: 0 | i32: 16448 | f32: 3.0 | fp: FixedPoint(0.0019607544)) }, Value { Bytes: [0, 0, 80, 40] (i: 0 | i32: 32832 | f32: 4.0 | fp: FixedPoint(0.0039138794)) }, Value { Bytes: [0, 0, a0, 40] (i: 0 | i32: 41024 | f32: 5.0 | fp: FixedPoint(0.004890442)) }] } (0x2085 | 8325) <> PEQ_1_10
[0.247s] Recv: []
[0.249s] Recv: Ack
```
The same problem happens also when I want to set the peq coefficients to the output: 
```
C:\workspace\minidsp-rs>minidsp -v --tcp 127.0.0.1 output 0 peq 0 set 1.0 -2.0 3.0 -4.0 5.0
error: Found argument '-2' which wasn't expected, or isn't valid in this context

  If you tried to supply '-2' as a value rather than a flag, use '-- -2'

Usage: minidsp output peq <INDEX> set [COEFF]...
```
### Help Text
The --help output is not very helpful as how to enter the values:
```
C:\workspace\minidsp-rs>minidsp input 0 peq 0 set --help
Set coefficients

Usage: minidsp input peq <INDEX> set [COEFF]...

Arguments:  [COEFF]...  Coefficients
```
The help text could be improved if it would indicate the order of the coefficients and how to format or separate them as in my example here:
```
Set coefficients

Usage: minidsp input peq <INDEX> set <b0> <b1> <b2> <a1> <a2> 

Arguments:
  <##> Biquad coefficient, decimal representation of float value, e. g. -0.123456
```
### Version
This was tested on Windows 11 with the mindsp version 0.1.9: 
```
C:\workspace\minidsp-rs>minidsp --version
minidsp 0.1.9

C:\workspace\minidsp-rs>ver

Microsoft Windows [Version 10.0.22621.2715]
```

mrene, 2023-11-28:
The parser library requires `--` to be before all positional arguments, all further arguments are then interpreted as values instead of as options.

In your case you can do `minidsp -v --tcp 127.0.0.1 input 0 peq 0 set -- 1.0 -2.0 3.0 4.0 -5.0`

hhartmut, 2023-11-28:
Thanks for the help. That works. 

I did try to put the `--` before the actual negative values but that did not work (see above)

So I think the help text could be improved to advise like in your example. Also it would be helpful to advise in the help text in which order the biquad coefficients have to be put. (I could figure that out myself by analyzing the debug output and comparing with a biquad file import, but as they say: explicit is better than implicit....) 

Anyhow, consider the case closed with my thanks for your effort and this great piece of software.

closed 2023-11-29 by hhartmut