mrene/minidsp-rs #657: MiniDSP SHD Power support

toini, 2024-02-12:
Hi! I have an SHD Power and would like to help with adding support for it (I do see SHD series is listed as full support but for now im able to only issue commands in the basic support category).

What exactly would be needed? Im unsure what is the "plugin application" mentioned in the devices page. (I also see there is a TODO item for "Add guide for adding new devices").

```
$ minidsp -vvv status
[0.000s] Sent: [31]
[0.000s] Sent: ReadHardwareId
[0.002s] Recv: [31, 01, 3f, 15, 00, 00, 00, 00, 00, 00, 00]
[0.002s] Recv: HardwareId(HardwareId { fw_major: 01, fw_minor: 3f, hw_id: 15 })
[0.002s] Sent: [05, ff, a1, 01]
[0.002s] Sent: ReadMemory { addr: ffa1, size: 01 }
[0.004s] Recv: [05, ff, a1, 69]
[0.004s] Recv: MemoryData(MemoryView { base: ffa1, data: [69] })
[0.004s] Sent: [05, ff, fe, 02]
[0.004s] Sent: ReadMemory { addr: fffe, size: 02 }
[0.006s] Recv: [05, ff, fe, 04, 13]
[0.006s] Recv: MemoryData(MemoryView { base: fffe, data: [04, 13] })
[0.006s] Sent: [05, ff, d8, 09]
[0.007s] Sent: ReadMemory { addr: ffd8, size: 09 }
[MasterStatus { preset: 1, source: NotInstalled, volume: Gain(-31.0), mute: false, dirac: false }
Input levels:
Output levels:
```

```
$ minidsp -vvv probe
[0.000s] Sent: [31]
[0.000s] Sent: ReadHardwareId
[0.002s] Recv: [31, 01, 3f, 15, 00, 00, 00, 00, 00, 00, 00]
[0.002s] Recv: HardwareId(HardwareId { fw_major: 01, fw_minor: 3f, hw_id: 15 })
[0.002s] Sent: [05, ff, a1, 01]
[0.002s] Sent: ReadMemory { addr: ffa1, size: 01 }
[0.004s] Recv: [05, ff, a1, 69]
[0.004s] Recv: MemoryData(MemoryView { base: ffa1, data: [69] })
[0.004s] Sent: [05, ff, fe, 02]
[0.004s] Sent: ReadMemory { addr: fffe, size: 02 }
[0.005s] Recv: [05, ff, fe, 04, 13]
[0.006s] Recv: MemoryData(MemoryView { base: fffe, data: [04, 13] })
0: Found Generic with serial 901043 at ws://localhost/devices/0/ws [hw_id: 21, dsp_version: 105]
```

This works:
```
$ minidsp mute on
```

But this does not:
```
$ minidsp -vvv output 0 mute on
[0.000s] Sent: [31]
[0.000s] Sent: ReadHardwareId
[0.001s] Recv: [31, 01, 3f, 15, 00, 00, 00, 00, 00, 00, 00]
[0.002s] Recv: HardwareId(HardwareId { fw_major: 01, fw_minor: 3f, hw_id: 15 })
[0.002s] Sent: [05, ff, a1, 01]
[0.002s] Sent: ReadMemory { addr: ffa1, size: 01 }
[0.003s] Recv: [05, ff, a1, 69]
[0.003s] Recv: MemoryData(MemoryView { base: ffa1, data: [69] })
[0.003s] Sent: [05, ff, fe, 02]
[0.003s] Sent: ReadMemory { addr: fffe, size: 02 }
[0.005s] Recv: [05, ff, fe, 04, 13]
[0.005s] Recv: MemoryData(MemoryView { base: fffe, data: [04, 13] })
Error: Specified channel or peq is out of range
```

toini, 2024-02-12:
I do see `hw_id` 15 is not being mapped atm. [SHD](https://github.com/mrene/minidsp-rs/blob/dev/protocol/src/device/probe.rs#L67) is 14. Is it safe to add that and try or do i risk breaking something?

toini, 2024-02-12:
Changing [this line](https://github.com/mrene/minidsp-rs/blob/dev/protocol/src/device/probe.rs#L67) to 
```
(14 | 21, _) => Shd,
```
Did the trick. Though i think i now lost my preset 2 when connecting. I then switched to preset 3 and i think it wants to do the same overwrite/reset to default always when connecting:
[image]
[image]

mrene, 2024-02-13:
Hi!

The plugin application was how devices were handled before the Device Console came out. As you figured out, you can force the device to be detected by changing the probe mapping.

You may want to check that the addresses for the SHD and SHD Power are indeed the same. Since the device console came out, there are now XML files that describe the addresses for each parameter, these files ship with the application.

I haven't had time to port minidsp-rs to use these XML files directly though. As for device damage I believe that most (all?) issues can be resolved by a device reset. Just keep in mind that you won't see your changes in the app, because it cannot read any parameters from the device itself (outside of some master settings).

toini, 2024-02-24:
I found an xml file in C:\Users\<me>\AppData\Local\Programs\minidsp-device-console\resources\device_support_package\info.xml on Windows.

```<XML>
	<description>
		This is the first file the app should read to see if need an update on of this list to support new devices.
		The following fields are currently being used by the Plugin
		- UDP_Listen
		- TCP_Connect
		- USB > DEVICE
		- DEVICES > HW
		- DEVICES > HW > DSP_STRUCTURES > DSP
	</description>
	
	<version>0.0.1</version>
	<timestamp>1234567</timestamp>
	<UDP_Listen>3999</UDP_Listen>
	<TCP_Connect>5333</TCP_Connect>
	<USB>
		<DEVICE vid="0x04D8" pid="0x003F" /> <!-- all PIC base product without DIRAC -->
		<DEVICE vid="0x2752" pid="0x0011" /> <!-- 2x4 HD / Flex -->
		<DEVICE vid="0x2752" pid="0x0013" /> <!-- HA-DSP -->
		<DEVICE vid="0x2752" pid="0x0040" /> <!-- DDRC-22 -->
		<DEVICE vid="0x2752" pid="0x0043" /> <!-- DDRC-88 -->
		<DEVICE vid="0x2752" pid="0x0044" /> <!-- DDRC-24 / Flex (DL) -->
		<DEVICE vid="0x2752" pid="0x0045" /> <!-- SHD -->
		<DEVICE vid="0x2752" pid="0x0047" /> <!-- CDSP-8x12/DL -->
		<DEVICE vid="0x2752" pid="0x0048" /> <!-- Harmony DSP -->
		<DEVICE vid="0x2752" pid="0x0012" /> <!-- miniDSP Flex Eight -->
		<DEVICE vid="0x2752" pid="0x004A" /> <!-- miniDSP Flex Eight (DL) -->
		<DEVICE vid="0x2752" pid="0x004B" /> <!-- miniDSP Flex HT / Flex HTx -->
	</USB>
	<DEVICES>
		<HW id="10" name="2x4 HD">
			<CTL_TYPE>XMOS</CTL_TYPE>
			<DSP_TYPE>ADI_SHARC_DSP</DSP_TYPE>
			<STORAGE>SPI_32Mb</STORAGE>
			<FW_TIMESTAMP>546454</FW_TIMESTAMP>
			<DSP_STRUCTURES>
				<DSP id="100" name="2x4 HD" intermediateId="200">
					<CMD_VERSION>3</CMD_VERSION>
					<DSP_TIMESTAMP>4564545</DSP_TIMESTAMP>
				</DSP>
				<DSP id="101" name="DDRC-24" intermediateId="201">
					<CMD_VERSION>3</CMD_VERSION>
					<DSP_TIMESTAMP>4564545</DSP_TIMESTAMP>
				</DSP>
			</DSP_STRUCTURES>			
		</HW>
		<HW id="27" name="Flex">
			<CTL_TYPE>XMOS</CTL_TYPE>
			<DSP_TYPE>ADI_SHARC_DSP</DSP_TYPE>
			<STORAGE>SPI_32Mb</STORAGE>
			<FW_TIMESTAMP>546454</FW_TIMESTAMP>
			<DSP_STRUCTURES>
				<DSP id="100" name="Flex" intermediateId="200">
					<CMD_VERSION>3</CMD_VERSION>
					<DSP_TIMESTAMP>4564545</DSP_TIMESTAMP>
				</DSP>
				<DSP id="101" name="Flex (DL)" intermediateId="201">
					<CMD_VERSION>3</CMD_VERSION>
					<DSP_TIMESTAMP>4564545</DSP_TIMESTAMP>
				</DSP>
			</DSP_STRUCTURES>			
		</HW>		
		<HW id="30" name="Flex Eight">
			<CTL_TYPE>XMOS</CTL_TYPE>
			<DSP_TYPE>ADI_SHARC_DSP</DSP_TYPE>
			<STORAGE>SPI_32Mb</STORAGE>
			<FW_TIMESTAMP>546454</FW_TIMESTAMP>
			<DSP_STRUCTURES>
				<DSP id="110" name="Flex Eight" intermediateId="210">
					<CMD_VERSION>3</CMD_VERSION>
					<DSP_TIMESTAMP>4564545</DSP_TIMESTAMP>
				</DSP>
				<DSP id="111" name="Flex Eight (DL)" intermediateId="211">
					<CMD_VERSION>3</CMD_VERSION>
					<DSP_TIMESTAMP>4564545</DSP_TIMESTAMP>
				</DSP>
			</DSP_STRUCTURES>			
		</HW>
		<HW id="31" name="Flex HT">
			<CTL_TYPE>XMOS</CTL_TYPE>
			<DSP_TYPE>ADI_SHARC_DSP</DSP_TYPE>
			<STORAGE>SPI_32Mb</STORAGE>
			<FW_TIMESTAMP>546454</FW_TIMESTAMP>
			<DSP_STRUCTURES>
				<DSP id="113">
					<CMD_VERSION>3</CMD_VERSION>
					<DSP_TIMESTAMP>4564545</DSP_TIMESTAMP>
				</DSP>
			</DSP_STRUCTURES>			
		</HW>
		<HW id="32" name="Flex HTx">
			<CTL_TYPE>XMOS</CTL_TYPE>
			<DSP_TYPE>ADI_SHARC_DSP</DSP_TYPE>
			<STORAGE>SPI_32Mb</STORAGE>
			<FW_TIMESTAMP>546454</FW_TIMESTAMP>
			<DSP_STRUCTURES>
				<DSP id="113">
					<CMD_VERSION>3</CMD_VERSION>
					<DSP_TIMESTAMP>4564545</DSP_TIMESTAMP>
				</DSP>
			</DSP_STRUCTURES>			
		</HW>
		<HW id="14" name="SHD">
			<CTL_TYPE>XMOS</CTL_TYPE>
			<DSP_TYPE>ADI_SHARC_DSP</DSP_TYPE>
			<STORAGE>SPI_32Mb</STORAGE>
			<FW_TIMESTAMP>546454</FW_TIMESTAMP>
			<DSP_STRUCTURES>
				<DSP id="105">
					<CMD_VERSION>3</CMD_VERSION>
					<DSP_TIMESTAMP>4564545</DSP_TIMESTAMP>
				</DSP>
			</DSP_STRUCTURES>			
		</HW>
		<HW id="17" name="SHD STUDIO">
			<CTL_TYPE>XMOS</CTL_TYPE>
			<DSP_TYPE>ADI_SHARC_DSP</DSP_TYPE>
			<STORAGE>SPI_32Mb</STORAGE>
			<FW_TIMESTAMP>546454</FW_TIMESTAMP>
			<DSP_STRUCTURES>
				<DSP id="105">
					<CMD_VERSION>3</CMD_VERSION>
					<DSP_TIMESTAMP>4564545</DSP_TIMESTAMP>
				</DSP>
			</DSP_STRUCTURES>			
		</HW>
		<HW id="21" name="SHD POWER">
			<CTL_TYPE>XMOS</CTL_TYPE>
			<DSP_TYPE>ADI_SHARC_DSP</DSP_TYPE>
			<STORAGE>SPI_32Mb</STORAGE>
			<FW_TIMESTAMP>546454</FW_TIMESTAMP>
			<DSP_STRUCTURES>
				<DSP id="105">
					<CMD_VERSION>3</CMD_VERSION>
					<DSP_TIMESTAMP>4564545</DSP_TIMESTAMP>
				</DSP>
			</DSP_STRUCTURES>			
		</HW>
	</DEVICES>
</XML>
```

It appears content in the HW elements for SHD, SHD STUDIO and SHD POWER do equal. Does that confirm compatibility in terms of addresses you mentioned?

bobmorane06, 2025-04-02:
Hi @toini, thanks for your posts. I just got an SHD Power and I'm trying to create automations in Home Assistant, so I've connected a rpi5 to the USB port of the SHD and I'm trying minidsp-rs.

I made your change `(14 | 21, _) => Shd,` and I can mute output 0, `config` and `dirac` commands work. However the selection of the source doesn't work (status reports `source: NotInstalled`, did you face that issue and solve it ? 

```
minidsp  status
MasterStatus { preset: 2, source: NotInstalled, volume: Gain(-27.0), mute: false, dirac: true }
Input levels: -120.0, -120.0
Output levels: -120.0, -120.0, -120.0, -120.0
```

bobmorane06, 2025-04-02:
I can change the source with a debug command, I found out with trial and error, with inspiration from @florentin in [this issue](https://github.com/mrene/minidsp-rs/issues/566#issuecomment-1475199022)

To set source = toslink, run:
`minidsp debug send 3400`

The different sources of the SHD Power: 
3400 toslink
3401 spdif
3402 aes
3403 usb
3404 lan

toini, 2025-04-02:
> did you face that issue and solve it ?

Good thing you seem to have figured it out. I never needed to switch source in my automation setup and actually have since switched to another device capabable of decoding surround sound.