mrene/minidsp-rs #663: Add ability to put device in "standby"

jprouty, 2024-06-05:
I am a newcomer - first off, thanks for the project! It looks very useful. Knowing the rigor of the DYI audio community and the shear issue count, this tool looks pretty mature.

I have a

danielhstahl, 2024-06-08:
I'd be very interested in this as well...I doubt the FlexHT supports triggers, but it may support Standby (though I've only seen IR codes for Standby, not TCP...https://support.minidsp.com/support/solutions/articles/47001137495-ir-commands-for-minidsp-products).  

I'd be happy to contribute to add this if anyone has docs on the TCP commands required for standby/on.

danielhstahl, 2024-06-30:
@jprouty not sure if this will work for you situation, but I successfully got my amp to turn off and on via the 12V trigger by using a USB->mono 1/8 inch jack from my RPI following this: https://www.audiosciencereview.com/forum/index.php?threads/usb-to-12v-trigger-input.19971/.  My code to implement this is here: https://github.com/danielhstahl/minidsp-remote-control/blob/main/server/index.js#L16-L28.  Seems hacky but has worked flawlessly so far.

pixeleyesltd, 2024-07-25:
I contacted miniDSP about this and they said it's only implemented at the IR level for the Flex models.

Jacob-Bishop, 2024-08-21:
For anyone else trying to solve the trigger problem: As an alternative to the Pi solution, I’ve had good luck using this Emotiva trigger hooked up to a smart plug https://emotiva.com/products/et-3 to turn on my amp.

There’s also a cheap Tuya Zigbee Smart IR remote on aliexpress (controllable via zigbee2mqtt or similar) that might work if you want to automate putting the minidsp device into standby via ir and triggering it that way?

mglobetx, 2025-04-02:
I worked around this by installing a TP-Link smart plug, and then plugging my MiniDSP Flex into that.  Then I set up the device I use as a remote to turn the smart plug on and off with the TP-Link API.   I'd rather toggle standby, but this does have the advantage of rebooting the Flex each time.  I have other devices in the house (my TV for one) that I reboot each night using a smart plug.  It helps kill most of the firmware gremlins.