mrene/minidsp-rs #670: Error with FlexHTx when addressing output 7

conrad-nash, 2024-10-27:
I'm experiencing an issue using minidsp-rs and a flexHTx. It looks like the final output isn't supported with the latest build.
I'm using minidsp-rs 0.1.10. 

If I attempt to mute output 7 I get a success message:

`pi@HTx:~ $ minidsp --force-kind flexhtx output 6 mute on`

That command works

If I try the same command for output 8 (index 7) I get an out of range error:

`pi@HTx:~ $ minidsp --force-kind flexhtx output 7 mute on
Error: Specified channel or peq is out of range`

@danielhstahl I believe you've been working in this area most recently, is this behavior you can confirm?

danielhstahl, 2024-10-27:
Hmm, odd that it wouldn't work as this shows output support for 0-7: https://github.com/mrene/minidsp-rs/blob/v0.1.10/devtools/src/codegen/flexhtx/mod.rs#L103

conrad-nash, 2024-10-27:
If I just run the minidsp command I get this:

`pi@HTx:~ $ minidsp --force-kind flexhtx
MasterStatus { preset: 0, source: Toslink, volume: Gain(-0.0), mute: false, dirac: false }
Input levels: -120.0, -120.0, -120.0, -120.0, -120.0, -120.0, -120.0
Output levels: -120.0, -120.0, -120.0, -120.0, -120.0, -120.0, -120.0`

Note that there's only 7 output levels, not 8.

conrad-nash, 2024-10-27:
Amazing!
I can test as soon as there's a build.

closed 2024-10-27 by mrene

conrad-nash, 2024-10-27:
Unless I'm being stupid this isn't working for me.

I still get 7 levels when I query minidsp, and I can't mute or unmute output 7.

```
pi@HTx:~ $ minidsp --force-kind flexhtx
MasterStatus { preset: 0, source: Toslink, volume: Gain(-0.0), mute: false, dirac: false }
Input levels: -120.0, -120.0, -120.0, -120.0, -120.0, -120.0, -120.0
Output levels: -120.0, -120.0, -120.0, -120.0, -120.0, -120.0, -120.0
```

```
pi@HTx:~ $ minidsp --version
minidsp 0.1.11
pi@HTx:~ $ minidsp --force-kind flexhtx output 7 mute on
Error: Specified channel or peq is out of range
```

It's been a long time since I've done any coding but the outputs section at line 1664 in flexthtx.rs appears to only have 7 addressable outputs, is that related?

mrene, 2024-10-27:
Oh I missed that only the codegen code was changed, but the code wasn't regenerated.

mrene, 2024-10-27:
I triggered another build with the regenerated code.

mrene, 2024-10-27:
@conrad-nash Try this one https://github.com/mrene/minidsp-rs/releases/tag/v0.1.12-pre1

conrad-nash, 2024-10-27:
That appears to have fixed it, thank you