mrene/minidsp-rs #687: Relative master gain change via API?

mglobetx, 2024-12-10:
I've been able to make relative gain setting changes from the command line, but when I run the daemon and communicate via the API, I only see a way to set the gain to a specific value.  Is the -r feature that's in the command line missing from the API?  I want to be able to send instructions to turn the master gain up or down 0.5db (or more) regardless of the value of the current gain setting.

bjornfor, 2024-12-30:
> I've been able to make relative gain setting changes from the command line, [...]

Can I ask how?

I was looking for JSON output on the CLI, but found none, and started making a script that extracted the current gain out of `minidsp status`, with the goal of making a relative gain adjustment command.

mrene, 2024-12-30:
> > I've been able to make relative gain setting changes from the command line, [...]
> 
> Can I ask how?
You can use `minidsp gain -r ...` 

```
❯ minidsp gain --help
Set the master output gain [-127, 0]

Usage: minidsp gain [OPTIONS] <VALUE>

Arguments:
  <VALUE>  Gain in decibels, between -127 and 0 inclusively

Options:
  -r, --relative  Specify the value as a relative increment on top of the current gain
  -h, --help      Print help
```

> 
> I was looking for JSON output on the CLI, but found none, and started making a script that extracted the current gain out of `minidsp status`, with the goal of making a relative gain adjustment command.

`minidsp -o json` will output JSON to stdout

--

bjornfor, 2024-12-30:
Thanks!

Apparently I'm terrible at reading help texts :see_no_evil:.

mglobetx, 2025-01-06:
I’ve finally been successful setting up a Xencels Lab Quick Keys to control my miniDSP Flex, which is connected to my Mac Mini by USB, and which serves as my Roon Server.  I ended up using Better Touch Tool to translate the dial rotation and key presses into command line instructions.  Working great. I can select profiles, inputs, mute, and most importantly to me control volume with the dial on the Quick Keys.