mrene/minidsp-rs #698: Device definition for Flex HT

kmarkley, 2025-01-30:
Would it be possible to have minidsp recognize the Flex HT model (in addition to the Flex HTx)?  I believe that except for having no analog input, the two units function identically.

```
~$ minidsp -vv probe
[0.000s] Sent: [31]
[0.000s] Sent: ReadHardwareId 
[0.001s] Recv: [31, 01, 4a, 1f, 02, 00, 00, 00, 00, 00, 00]
[0.001s] Recv: HardwareId(HardwareId { fw_major: 01, fw_minor: 4a, hw_id: 1f })
[0.001s] Sent: [05, ff, a1, 01]
[0.001s] Sent: ReadMemory { addr: ffa1, size: 01 } 
[0.003s] Recv: [05, ff, a1, 71]
[0.003s] Recv: MemoryData(MemoryView { base: ffa1, data: [71] })
[0.003s] Sent: [05, ff, fe, 02]
[0.003s] Sent: ReadMemory { addr: fffe, size: 02 } 
[0.004s] Recv: [05, ff, fe, 04, 95]
[0.004s] Recv: MemoryData(MemoryView { base: fffe, data: [04, 95] })
0: Found Generic with serial 901173 at usb:3-1%3A1.3?vid=2752&pid=004b [hw_id: 31, dsp_version: 113]
```

Basic control (config, volume) works ok as generic, but it is impossible to change source and no input or output levels are reported.

Thanks

madsi1m, 2025-09-19:
I also would like the same.
Looking at the code we might be able to get away with adding `(31, 113) => FlexHtx` in https://github.com/mrene/minidsp-rs/blob/985ebd432813bbfae4a3ee02d24cadfbdf06e24c/protocol/src/device/probe.rs#L84 I have a feeling they are the same device inside

madsi1m, 2025-09-19:
actually they are different, i totally forgot about the analogue inputs

tiprya, 2026-09-27:
Not sure if it's helpful but I connected AI to my Flex HT and got it to add definitions to the minidsp-rs code.

It works perfectly for me and you're welcome to try it out, but fair warning that it's entirely AI-generated so I don't think appropriate for a PR: [https://github.com/tiprya/minidsp-rs/tree/flexht](https://github.com/tiprya/minidsp-rs/tree/flexht)