mrene/minidsp-rs #701: Support for new HTx architecture

conrad-nash, 2025-03-11:
Minidsp have released a new version of the firmware for the HTx which enables input PEQ. However, this version breaks the existing integration with minidsp-rs. I assume this is because of hierarchy change in the json, but I don't know for sure.

@danielhstahl I wondered whether this would be something you'd be interested in taking a look at? I'm happy to provide any information that I can as I have an HTx running the latest beta firmware. I'm no dev but I can follow instructions :)

danielhstahl, 2025-03-12:
Hmm, do you have any documentation on the beta firmware?  I'm still on stable (1.11).

conrad-nash, 2025-03-12:
No docs and minidsp are not keen to share.
I can definitely share the xml version file from the new device console and I wondered if there’s a way to inspect the commands as they pass “through” minidsp-rs from the native app?